microsoft/Azure-Kinect-Sensor-SDK #1064: GSTREAMER EXAMPLE

barakooda, 2020-02-06:
Hi all,
Anyone was able to convert the depth stream into png using 
gstreamer, similar to the example with the ffmpeg  ?

thanks,
B.

wes-b, 2020-02-20:
Does it need to be gstreamer? We have an example using OpenCV

```
static void image_to_file(const k4a::image &im, char *file_path, int image_num)
{
    if (!im.is_valid())
    {
        return;
    }

    char buffer[1024];
    snprintf(buffer, sizeof(buffer), "%s\\%03i_img.png", file_path, image_num);

    // https://docs.opencv.org/trunk/d4/da8/group__imgcodecs.html#gabbc7ef1aa2edfaa87772f1202d67e0ce

    cv::Mat im_mat;
    if (im.get_width_pixels() == 1280 || im.get_height_pixels() == 720)
    {
        // Color Image
        cv::Mat im_mat_temp(im.get_height_pixels(), im.get_width_pixels(), CV_8UC4, (void *)im.get_buffer());
        im_mat = im_mat_temp;
    }
    else
    {
        // Depth Image
        cv::Mat im_mat_grey(im.get_height_pixels(),
                            im.get_width_pixels(),
                            CV_16U,
                            (void *)im.get_buffer(),
                            static_cast<size_t>(im.get_stride_bytes()));

        uint32_t min = 250; 
        uint32_t max = 2500;
        for (int x = 0; x < im_mat_grey.cols; x++)
        {
            for (int y = 0; y < im_mat_grey.rows; y++)
            {
                ushort depth = im_mat_grey.at<ushort>(y, x);
                if (depth < min)
                {
                    depth = 0;
                }
                else if (depth > max)
                {
                    depth = 0xFFFF;
                }
                else
                {
                    uint32_t depth32 = (depth - min) * 0xFFFF / (max - min);
                    CV_Assert(depth32 <= 0xFFFF);
                    depth = (ushort)(depth32);
                }
                im_mat_grey.at<ushort>(y, x) = depth;
            }
        }

        cv::cvtColor(im_mat_grey, im_mat, cv::COLOR_GRAY2BGRA);
    }

    CV_Assert(im_mat.channels() == 4);

    vector<int> compression_params;
    compression_params.push_back(cv::IMWRITE_PNG_COMPRESSION);
    compression_params.push_back(9);

    bool result = false;
    try
    {
        result = imwrite(buffer, im_mat, compression_params);
    }
    catch (const cv::Exception &ex)
    {
        cerr << "Exception converting image to PNG format: " << ex.what();
        exit(1);
    }
    
    if (result)
        printf("Saved PNG file with alpha data.\n");
    else
        printf("ERROR: Can't save PNG file.\n");
}
```

closed 2020-02-20 by wes-b

AndreV84, 2021-05-27:
anyone was able to open kinect DK from gstreamer? is there any example of the pipeline?
was-b is there an example opencv opening kinect DK video stream? Typically I would open usb camera using gstreamer v4l driver , but how to open RGB Kinect from opencv?

AndreV84, 2021-05-27:
@wes-b do you have opencv video example that we could adopty into the file to open the video stream with it? https://github.com/AndreV84/mediapipe/blob/master/demo_run_graph_main_gpu_mod.cc_updated
https://github.com/google/mediapipe/blob/master/mediapipe/examples/desktop/demo_run_graph_main_gpu.cc