microsoft/Azure-Kinect-Sensor-SDK #1065: Conversion between C handle and CPP wrapper?

SeanStephensenAIRM, 2020-02-07:
I know this is probably a more general C/CPP question, but I'm having trouble finding what I'm looking for online. I began writing my program from [The example startup code on the Microsoft website](https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-app) and have developed that into a fairly large work using the same C based syntax and handles (e.g. `k4a_transformation_t`). A new functionality that I'd like to implement is very close to [the Green Screen example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen) which is written for CPP (e.g. `k4a::transformation`). 

In short, I want the green screen script to return a transformation handle (which is returned as a `k4a::transformation`) to my main script (which needs to accept a `k4a_transformation_t`). Of course, I'm getting the error no suitable conversion from `k4a::transformation to k4a_transformation_t`. Is there a simple way to convert between these?

Since green screen does almost exactly what I need, it would be nice to not have to refactor it into an identical C script, but as it is, the handles are incompatible. Probably obvious to anyone who's more experienced in these languages, but simply changing the handles does not work - the syntax is also different. I notice k4a::image has ::handle() which returns the underlying k4a_image_t, but transformation does not seem to have the same member.

wes-b, 2020-02-07:
It is not currently exposed by the CPP interface because there isn't an API outside of the transformation surface that needs it. Can you explain why we need access to it?

SeanStephensenAIRM, 2020-02-07:
I have a main code written in C syntax that captures, saves, shows, etc images through an interactive console app. It was written with one camera in mind, but now we're looking at adding multi-camera functionality. At the start of this program, cameras are opened and setup, including determining intrinsic transformations. My idea is to, during camera setup in the main program, add the functionality to either live calibrate these multiple cameras, or load in an existing calibration file (I'll figure that out later). First to live calibrate, I'd like to call the green screen program that does a live chessboard calibration. Instead of continuing on to produce a green screen, I'd like to just use that calculated extrinsic calibration to continue back into my main capture program and output these extrinsically transformed images.

In [the last line that I intend to use](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/f21aac905c4e8d416ae3dc4055f673d12a3ebaa7/examples/green_screen/main.cpp#L234) from the green screen code (the one that finally determines the extrinsic transformation handle), I realize I could also produce a k4a_transformation_t here if I had the extrinsic calibration as a k4a_calibration_t instead of a k4a::calibration, but I see there is also no member to reveal the underlying calibration handle.

wes-b, 2020-02-10:
Sorry I still don't follow. If we did expose a k4a_transformation_t handle, how would you use it?

SeanStephensenAIRM, 2020-02-10:
@wes-b as an argument for [k4a_transformation_depth_image_to_color_camera()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gafacffb5f781a9c2df30d4a16241cd514.html) which is already implemented in my program.

wes-b, 2020-02-11:
Gotcha, our API isn't really designed to bounce between C and CPP. The expectation being that a user will use one or the other, but not both. 

With that being said it would not be that hard to build your own version and add that functionality.

You might also consider converting your CPP code to C. It should be fine in a CPP project and then you have access to the structures you need.

closed 2020-02-11 by wes-b