microsoft/Azure-Kinect-Sensor-SDK #1067: Body Tracker Error: Failed to enqueue capture to tracker!

petrosferdinand, 2020-02-10:
Hi! I'm having trouble reading some images from mkv files. When I try to enable body tracking, I get the following error:

[2020-02-10 18:02:21.467] [error] [t=8883] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2020-02-10 18:02:21.468] [error] [t=8883] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (359): Invalid argument to image_get_buffer(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2020-02-10 18:02:21.468] [error] [t=8883] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/DepthFrameBlobK4A.cpp (19): Initialize(). Get IR buffer from the capture handle failed!
[2020-02-10 18:02:21.468] [error] [t=8883] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/TrackerHost.cpp (247): EnqueueCapture(). Initialize DepthFrameBlob failed!
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to enqueue capture to tracker!
[kinect_0/node-4] process has died [pid 8824, exit code -6, cmd /root/catkin_ws/devel/lib/azure_kinect_ros_driver/node __name:=node __log:=/root/.ros/log/260495b0-4c2f-11ea-a7bc-0242ac110002/kinect_0-node-4.log].
log file: /root/.ros/log/260495b0-4c2f-11ea-a7bc-0242ac110002/kinect_0-node-4*.log

I'm using the Azure_Kinect_ROS_Driver, but I'm posting it here, since I believe that the error is at the interface with the K4ABT.

wes-b, 2020-02-10:
Seems like you provided an invalid handle to one of our API's.  Can you share your code?

petrosferdinand, 2020-02-10:
like I said, I'm using the Azure_Kinect_ROS_Driver, specifically this branch: https://github.com/RoseFlunder/Azure_Kinect_ROS_Driver/tree/feat/bump-sdk-versions-1.3-1.0

I assume that the call happens here: 
https://github.com/RoseFlunder/Azure_Kinect_ROS_Driver/blob/feat/bump-sdk-versions-1.3-1.0/src/k4a_ros_device.cpp#L972

wes-b, 2020-02-20:
#1068

closed 2020-02-20 by wes-b