microsoft/Azure-Kinect-Sensor-SDK #1068: errors when using k4aBT and K4AROS on recorded mkv viles

petrosferdinand, 2020-02-10:
Hi! I'm having trouble reading some images from mkv files that were recorded with k4arecorder. When I try to enable body tracking, I get the following error:

[2020-02-10 18:02:21.467] [error] [t=8883] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2020-02-10 18:02:21.468] [error] [t=8883] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (359): Invalid argument to image_get_buffer(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2020-02-10 18:02:21.468] [error] [t=8883] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/DepthFrameBlobK4A.cpp (19): Initialize(). Get IR buffer from the capture handle failed!
[2020-02-10 18:02:21.468] [error] [t=8883] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/TrackerHost.cpp (247): EnqueueCapture(). Initialize DepthFrameBlob failed!
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to enqueue capture to tracker!
[kinect_0/node-4] process has died [pid 8824, exit code -6, cmd /root/catkin_ws/devel/lib/azure_kinect_ros_driver/node __name:=node __log:=/root/.ros/log/260495b0-4c2f-11ea-a7bc-0242ac110002/kinect_0-node-4.log].
log file: /root/.ros/log/260495b0-4c2f-11ea-a7bc-0242ac110002/kinect_0-node-4*.log

I'm using the Azure_Kinect_ROS_Driver, but I'm posting it here, since I believe that the error is at the interface with the K4ABT.

I recorded these files with k4arecorder as mkv's and I'm using the ROS interface to get the tracked bodies.

petrosferdinand, 2020-02-10:
on other files I get the following error:

`[2020-01-22 13:29:07.149] [error] [t=4073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/record/internal/matroska_read.cpp (1754): convert_block_to_image(). Failed to decompress jpeg image to BGRA format.
[2020-01-22 13:29:07.149] [error] [t=4073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/record/internal/matroska_read.cpp (1893): convert_block_to_image(context, block, &image_handle, context->color_format_conversion) returned failure in new_capture()
[2020-01-22 13:29:07.149] [error] [t=4073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/record/internal/matroska_read.cpp (2075): new_capture(context, blocks[i]->current_block.get(), capture_handle) returned failure in get_capture()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to get next capture!`

wes-b, 2020-02-20:
You need to debug the line of code that is generating this error:
```Invalid k4a_image_t (nil)```

From there you need to figure out why the image is NULL. Share your source and where the NULL image is appearing and we might be able to help.

closed 2020-02-20 by wes-b