microsoft/Azure-Kinect-Sensor-SDK #1070: Could you direct me 1.6.104.bin version of firmware?

spk921, 2020-02-10:
I know this might be the right place to post the questions.
I am seeing people using 1.6.104 version of firmware.
However, I only can see 1.6.102 version whenever I download the SDK also I was not able to find on the web. Could someone help on this issue? Or share the link?

I am using OS as Window10.

wes-b, 2020-02-10:
We did not release that version due to regressions. We published a 1.6.107 in an alpha package but that too has regressions on Linux so it has been unlisted. We are working fixes.

https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.2

closed 2020-02-10 by wes-b

spk921, 2020-02-11:
@wes-b  Thank you for the comments. How can I download the bin file? I am not familiar with the nuget. Could you help or link how I could download the file from your link above?

RoseFlunder, 2020-02-11:
@spk921 In your case you can just download the file on the linked website.
You can find "Download package" link on the right side in the Info section.
The downloaded file can be extracted like a zip file and it will contain a "firmware"-folder.
So you can just add a ".zip" file ending and extract it with the option in the windows explorer or extract it manually with a zip compatible extraction software of your choice (i.e. 7zip).