microsoft/Azure-Kinect-Sensor-SDK #1072: CMake Error at CMakeLists.txt

Elenaegr, 2020-02-11:
I want to generate a project for the green_screen example. When I generate the project from the CMake GUI with the following parameters:

**Where is the source code:** C:/Users/egneeol/Desktop/IMT Thesis/git/Azure-Kinect-Sensor-SDK/examples/green_screen
**Where to build the binaries:** C:/Users/egneeol/Desktop/IMT Thesis/git/Azure-Kinect-Sensor-SDK/build

And I added and entry called **OpenCV_LIBS** with the following path: C:/Users/egneeol/Desktop/IMT Thesis/git/Azure-Kinect-Sensor-SDK/build/Win-x64-Debug-Ninja/src/sdk/k4a.lib

However, when I generate the code, I get the following error:

CMake Error at CMakeLists.txt:4 (add_executable):
  Target "green_screen" links to target "k4a::k4a" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

Any help would be much appreciated.

Elenaegr, 2020-02-11:
The CMakeLists.txt looks exactly like the one I cloned from the source code:

`# Copyright (c) Microsoft Corporation. All rights reserved.
# Licensed under the MIT License.
add_executable(green_screen main.cpp)
target_include_directories( green_screen PRIVATE ${OpenCV_INCLUDE_DIRS} )
target_link_libraries(green_screen PRIVATE k4a::k4a ${OpenCV_LIBS})`

wes-b, 2020-02-11:
You also need logic to find OpenCV. Check out https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/cmake/FindOpenCV.cmake

Elenaegr, 2020-02-11:
Yes, the OpenCV_DIR variable was not being properly assigned. 
Already solved it. 
Thank you for yor help.

closed 2020-02-11 by wes-b