microsoft/Azure-Kinect-Sensor-SDK #1073: Cannot start camera when connected to power + data from single USB-C to USB-C cable

G-mel, 2020-02-12:
**Describe the bug**
After purchasing the following [usb-c cable](https://www.ntcdistributing.com/usb-3-1-type-c/c-to-c/usb-3-1-gen-2-c-male-to-c-male-cable/?sku=U4S5C01C01-030), which a sales rep confirmed works with Azure Kinect for data and power simultaneously, I get the following error when trying to run start the depth camera + rgb camera with `k4aviewer`.

```
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (737): usb_cmd_io(). Error calling libusb_bulk_transfer for status, result:LIBUSB_ERROR_TIMEOUT
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (799): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (389): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_NV_DATA_GET, (uint8_t *)&nv_tag, sizeof(nv_tag), calibration, cal_size, bytes_read) returned failure in depthmcu_get_cal()
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (439): depthmcu_get_cal(depth->depthmcu, depth->calibration_memory, DEPTH_CALIBRATION_DATA_SIZE, &depth->calibration_memory_size) returned failure in depth_start()
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-02-12 18:38:46.761] [error] [t=14823] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
```

The azure kinect is found and shows up with `lsusb`. The white LED on the kinect is solid white. My PC is currently using an Asrock B450 Steel Legend motherboard that has a single USB-C 3.1 Gen2 port.

**To Reproduce**
1. Plug in cable to kinect and PC
2. Run k4aviewer
3. Open device
4. Start RGB and Depth cameras

**Expected behavior**
Cameras start as usual

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.3.0
 - Firmware version: 
```Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft```

tesych, 2020-02-13:
@G-mel , have you tried using the out of the box cable that comes with the device? Do you get any problems?

G-mel, 2020-02-13:
@tesych  It works perfectly with the out of the box cables

PierrePlantard, 2020-02-13:
@G-mel  Several parameters of the cable you need to take into account.
The USB-C cable need to be power delivery and data transfer. 
The length of the USB-C is a limitation, please see the tips in https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification 
"A passive cable should be less than 1.5m in length. If longer, use an active cable."
It also depends of the power delivery capacity and data transfer speed.
In our test, the kinect azure works with : 
-  2m cable with Power : 60W and 10 Gbps data transfer
- 3m cable with Power : 100W and 20 Gbps data transfer
I hope this will help you

G-mel, 2020-02-13:
@PierrePlantard Thank you for the info. So the 3m cable I got is a passive cable with 5Gbps data transfer and meets the 1.5A power requirement as highlighted in the hardware requirements you linked. Seems as though the length and data transfer rate are the bottlenecks. I'll exchange the cable I got for one that meets the specs.

closed 2020-02-13 by G-mel