microsoft/Azure-Kinect-Sensor-SDK #1074: Body orientation flip when leaning backward

cavisions, 2020-02-13:
**Describe the bug**

When leaning backward the body orientation is flipping as visible on the picture below. It changes its state several times per second. Even when body is not flipped the detection quality is very bad as visible on the second picture. 

**To Reproduce**

This problem can be observed in the viewer application delivered with the body tracking SDK.

**Expected behavior**

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: 1903
 - SDK Version:  1.3.0
-  Body Tracking SDK Version: 1.0
 - Firmware version:  1.6

qm13, 2020-02-19:
I have been unable to replicate this issue.

What mode are you running the depth sensor?
Can you try to replicate with different clothing?
Can you try to replicate in a different environment (point camera in different direction or move setup to different room)?

qm13, 2020-03-11:
Closing as duplicate of #575 
@Chrisbright10 do you have a recording which demonstrates the issue? If so please provide me a contact email address and I can provide details on how to share it with Microsoft.

closed 2020-03-11 by qm13