microsoft/Azure-Kinect-Sensor-SDK #1078: When the KinectDK ran long time more than 3 days, it is going to be dead.

yeongwoonIm, 2020-02-17:
I've tested KinectDK long time stability while 3 days. and KinectDK is dead at  3rd day.
I got this error message on WinEvent Viewer
error logs ==>
설명: 처리되지 않은 예외로 인해 프로세스가 종료되었습니다.
예외 정보:System.NullReferenceException
   위치: Microsoft.Azure.Kinect.Sensor.LargeArrayPool.Rent(Int32)
   위치: Microsoft.Azure.Kinect.Sensor.Allocator.AllocateFunction(Int32, IntPtr ByRef)

translate ==>
Description:The process was terminated due to an unhandled exception
Exception Info:System.NullReferenceException
   at: Microsoft.Azure.Kinect.Sensor.LargeArrayPool.Rent(Int32)
   at: Microsoft.Azure.Kinect.Sensor.Allocator.AllocateFunction(Int32, IntPtr ByRef)

I checked Issues. but i doesn't helpful for me..
[https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/754](url)

is there avoiding errors?? or should I wait updated SDK??

the test Code is below.

Test Code is C#.
==> get data part.
            Capture capture;
            while (m_runflag)
            {
                try
                {
                    
                    capture = device.GetCapture();

                    DateTime StartDate = DateTime.Now;

                    depth = capture.Depth.GetPixels<short>().ToArray();
                    activeIR = capture.IR.GetPixels<short>().ToArray();
                    if (depth != null && activeIR != null)
                    {
                        IRDeviceData iRDeviceData = new IRDeviceData();
                        iRDeviceData.frameTime = StartDate;
                        iRDeviceData.depth = new short[depth.Length];
                        iRDeviceData.activeIR = new short[activeIR.Length];

                        Array.Copy(depth, iRDeviceData.depth, depth.Length);
                        Array.Copy(activeIR, iRDeviceData.activeIR, activeIR.Length);

                        IRDeviceDataCQ.Enqueue(iRDeviceData);
                    }
                }
                catch (Exception ex)
                {
                    logger.ErrorFormat("{0}", ex);
                    logger.ErrorFormat("{0}", ex.Message);
                }
                Thread.Sleep(1);
            }

==> processing data part
            while (m_runflag)
            {
                try
                {
                    if(IRDeviceDataCQ.IsEmpty == true)
                    {
                        Thread.Sleep(1);
                        continue;
                    }

                    IRDeviceData iRDeviceData;

                    if(IRDeviceDataCQ.TryDequeue(out iRDeviceData) == false)
                    {
                        Thread.Sleep(1);
                        continue;
                    }
                    //Do something.....
                    
                }
                catch (Exception ex)
                {
                    logger.ErrorFormat("{0}", ex);
                    logger.ErrorFormat("{0}", ex.Message);
                }
            }

PC
OS : windows 10 enterprise 2016 LTSB 64bit
CPU : intel i7-7700K cpu @ 3.60GHz
RAM : 8 GB
GPU : Intel HD630

SDK : 1.3.0

nicobeek, 2020-03-02:
I'm having the same issue: NullReferenceException on LargeArrayPool. Seems like the issue has been fixed, but the SDK hasn't been updated. Is there a new SDK installer with this fix included available?

yeongwoonIm, 2020-03-04:
> I'm having the same issue: NullReferenceException on LargeArrayPool. Seems like the issue has been fixed, but the SDK hasn't been updated. Is there a new SDK installer with this fix included available?

good to know~ thanks~
I can wait a new SDK Installer

closed 2020-03-04 by yeongwoonIm

wes-b, 2020-03-04:
> Is there a new SDK installer with this fix included available?

Hopefully in the next week or 2