microsoft/Azure-Kinect-Sensor-SDK #1079: K4aviewer : core dumped

Pavankv92, 2020-02-17:
Hi, I have error when I try to launch k4aviewer 

error : Segmentation fault (core dumped)
-----------------------------------------------------------
output file when I launch the ros driver for azure is below.

SUMMARY
========

PARAMETERS
 * /azure_kinect_ros_driver/body_tracking_enabled: False
 * /azure_kinect_ros_driver/body_tracking_smoothing_factor: 0.0
 * /azure_kinect_ros_driver/color_enabled: True
 * /azure_kinect_ros_driver/color_format: bgra
 * /azure_kinect_ros_driver/color_resolution: 1536P
 * /azure_kinect_ros_driver/depth_enabled: True
 * /azure_kinect_ros_driver/depth_mode: WFOV_UNBINNED
 * /azure_kinect_ros_driver/fps: 15
 * /azure_kinect_ros_driver/imu_rate_target: 0
 * /azure_kinect_ros_driver/ir_mono8_scaling_factor: 1.0
 * /azure_kinect_ros_driver/point_cloud: True
 * /azure_kinect_ros_driver/point_cloud_in_depth_frame: False
 * /azure_kinect_ros_driver/recording_file: 
 * /azure_kinect_ros_driver/recording_loop_enabled: False
 * /azure_kinect_ros_driver/rescale_ir_to_mono8: False
 * /azure_kinect_ros_driver/rgb_point_cloud: True
 * /azure_kinect_ros_driver/sensor_sn: 
 * /azure_kinect_ros_driver/tf_prefix: 
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    azure_kinect_ros_driver (azure_kinect_ros_driver/node)
    joint_state_publisher_azure (joint_state_publisher/joint_state_publisher)
    robot_state_publisher_azure (robot_state_publisher/state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[joint_state_publisher_azure-1]: started with pid [29568]
process[robot_state_publisher_azure-2]: started with pid [29569]
process[azure_kinect_ros_driver-3]: started with pid [29570]
[ WARN] [1581951372.891065940]: The 'state_publisher' executable is deprecated. Please use 'robot_state_publisher' instead
[ INFO] [1581951373.004918358]: K4A Parameters:
[ INFO] [1581951373.005387666]: sensor_sn - std::string : 
[ INFO] [1581951373.005404353]: depth_enabled - bool : 1
[ INFO] [1581951373.005419870]: depth_mode - std::string : WFOV_UNBINNED
[ INFO] [1581951373.005439186]: color_enabled - bool : 1
[ INFO] [1581951373.005457178]: color_format - std::string : bgra
[ INFO] [1581951373.005475190]: color_resolution - std::string : 1536P
[ INFO] [1581951373.005503903]: fps - int : 15
[ INFO] [1581951373.005525254]: point_cloud - bool : 1
[ INFO] [1581951373.005545151]: rgb_point_cloud - bool : 1
[ INFO] [1581951373.005566291]: point_cloud_in_depth_frame - bool : 0
[ INFO] [1581951373.005582868]: tf_prefix - std::string : 
[ INFO] [1581951373.005602113]: recording_file - std::string : 
[ INFO] [1581951373.005619396]: recording_loop_enabled - bool : 0
[ INFO] [1581951373.005638242]: body_tracking_enabled - bool : 0
[ INFO] [1581951373.005671535]: body_tracking_smoothing_factor - float : 0
[ INFO] [1581951373.005688881]: imu_rate_target - int : 0
[ INFO] [1581951373.005708131]: rescale_ir_to_mono8 - bool : 0
[ INFO] [1581951373.005727459]: ir_mono8_scaling_factor - float : 1
[ INFO] [1581951373.009676714]: Found 1 sensors
[ INFO] [1581951373.009709994]: No serial number provided: picking first sensor
[ INFO] [1581951373.211264129]: K4A[0] : 000328192012
[ INFO] [1581951373.261679246]: K4A Serial Number: 000328192012
[ INFO] [1581951373.267142793]: RGB Version: 1.6.102
[ INFO] [1581951373.267175221]: Depth Version: 1.6.75
[ INFO] [1581951373.267187172]: Audio Version: 1.6.14
[... 86 lines cut ...]
[ INFO] [1581951374.171371958]: 		 Extrinsics:
[ INFO] [1581951374.171396466]: 			 Translation: -46.9498, -28.614, 3.49653
[ INFO] [1581951374.171416273]: 			 Rotation[0]: 0.00137554, 0.0184903, -0.999828
[ INFO] [1581951374.171435780]: 			 Rotation[1]: -0.999983, -0.00570376, -0.00148123
[ INFO] [1581951374.171455264]: 			 Rotation[2]: -0.00573017, 0.999813, 0.0184822
[ INFO] [1581951374.171472166]: 	 IMU (IMU to Color):
[ INFO] [1581951374.171489552]: 		 Extrinsics:
[ INFO] [1581951374.171512884]: 			 Translation: -28.5289, -2.79097, -47.0488
[ INFO] [1581951374.171534393]: 			 Rotation[0]: 0.00137554, -0.999983, -0.00573017
[ INFO] [1581951374.171554518]: 			 Rotation[1]: 0.0184903, -0.00570376, 0.999813
[ INFO] [1581951374.171576166]: 			 Rotation[2]: -0.999828, -0.00148123, 0.0184822
[ INFO] [1581951374.171659647]: STARTING CAMERAS
[azure_kinect_ros_driver-3] process has died [pid 29570, exit code -11, cmd /home/Vishwanath/catkin_ws/devel/lib/azure_kinect_ros_driver/node __name:=azure_kinect_ros_driver __log:=/home/Vishwanath/.ros/log/98ac35c8-5193-11ea-9d58-1866da36604f/azure_kinect_ros_driver-3.log].
log file: /home/Vishwanath/.ros/log/98ac35c8-5193-11ea-9d58-1866da36604f/azure_kinect_ros_driver-3*.log

----------------------------------------------------------------------------------------------
system config 
ubuntu 18.04, OpenGL version string: 4.6.0 NVIDIA 440.33.01
-------------------------------------------------------------------------------------------------

I have used azure in the same system few days ago and everything was fine. but now i get this error.
any help is appreciated.

Thanks
Pavan

wes-b, 2020-02-20:
Hello @Pavankv92, please in the future use our bug template to answer basic questions to about issue being opened (for most it helps reduce back and forth about the issue). For example what are the steps for reproducing. As just running the viewer on my Linux PC won't reproduce this issue.

Any chance you are trying to use the machine remotely?

closed 2020-03-02 by wes-b

zapaishchykova, 2021-06-15:
hey @wes-b, why is it a problem to use a machine remotely? I have the same error with Jetson Nano headless mode and Azure Kinekts, and this error occurs completely randomly