microsoft/Azure-Kinect-Sensor-SDK #1081: Get the raw color image without distortion correction

etiennedub, 2020-02-18:
Hi,
From my experimentation, I conclude that the color image return by `k4a_capture_get_color_image` has distortion correction applied from the internal intrisincs parameters.

I have two questions related to this: 
1. Is it possible to get the raw color image without correction ?
2. Where is applied the correction in the code and how could I overwrite those intrinsics parameters ?

Thank you

rabbitdaxi, 2020-02-20:
@etiennedub the sdk does not do undistortion naively. The API you are refering to will give you the raw color image without any distortion correction

etiennedub, 2020-02-20:
Ok that's weird. I did a quick experiment to test if the color image was undistort. I simply call `k4a_transformation_depth_image_to_color_camera` two times with different distorsion parameters for the color camera. One time with the default parameters and the second time all the parameters are set to 0.
The results are that the depth is better align to the color image with the default distorsion coefficents. So, I concluded that the color image was undistor. I think I was wrong in my conclusion and the color camera distortion is apply to the reproject depth instead. The depth is distor instead of the color undistor. Am I right ?
 
You can see the result here. The one with the worst aligment is the one with all distorsion coefficients set to 0.
[image]

rabbitdaxi, 2020-02-20:
@etiennedub both color and depth image are distorted from the SDK. 
When you call k4a_transformation_depth_image_to_color_camera, it essentially will unprojectthe depth image to depth point cloud in depth camera space, then transformed to color camera space, then project to the distorted color image to align with the color. Therefore, this explains your observation of "The results are that the depth is better align to the color image with the default distorsion coefficents."

closed 2020-02-21 by etiennedub

tanyouliang95, 2020-05-03:
Hi, I have one quick question regards the distortion coefficient. As u mentioned here, the distortion is not explicitly shown, I suppose it is being processed internally (in the function that you mentioned) during rectification to the `camera_frame`. As such, the `depth_image_descriptor` has specified all the required intrinsic parameters, and undistort action is being done internally. Correct me if I am wrong with this. 
Thanks