microsoft/Azure-Kinect-Sensor-SDK #1082: Body Index Map not properly transformed to colour camera

eshirima, 2020-02-18:
**Describe the bug**
Upon attaining the body index map from the body tracker, I convert it to the colour camera as shown [here](https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/camera_space_transform_sample/main.cpp#L58). Thereafter I go ahead and draw it out on the colour image for verification purposes but their seems to be an issue with the aspect ratio or some sort of offset since the map is never aligned to the person in the colour camera.

This is the body index map as seen from the colour camera space
[image]

But when I draw it directly on the colour image, this is the result
[image]

The problem becomes even more apparent when the body walks outside the depth camera's FOV as seen from the body index map in colour space shown below
[image]

and when drawn directly on the colour image, this is what I get
[image]

From this, it is evident that the body index map is not fully/properly transformed to the colour camera. Any ideas on how to resolve this?

I am using version 1.0.0 of the Body Tracking SDK with the following configurations

_Colour Camera_: `K4A_IMAGE_FORMAT_COLOR_MJPG`, `K4A_COLOR_RESOLUTION_1080P`
_Depth Camera_: `K4A_DEPTH_MODE_NFOV_UNBINNED`

**PS:** _I fully utilised the code samples for transformations and simply converted them to OpenCV Mat for visualisation_

yijiew, 2020-02-20:
Hi @eshirima , thanks for posting this issue. When you draw the body tracking result on your color image, I will expect at least you won't see this color strip artifact... There seems to be something wrong with your drawing code instead of the body index mask. Could you share the code you use to draw the body index map?

eshirima, 2020-02-20:
@yijiew Thanks for the rather quick response. The artifact on the colour image is simply me colouring the mapped regions with a different pixel colour for differentiation as shown below. `indexMap` is the produced index map in colour camera space and `colour` is the raw RGBA image from the sensor.

```
Mat drawIndexMapOnColourImage(Mat indexMap, Mat colour)
{
	Mat result;
	colour.copyTo(result);

	for (size_t row = 0; row < indexMap.rows; ++row)
	{
		// grab first pixel in the row
		uchar* currentPixel = indexMap.ptr<uchar>(row);

		for (size_t col = 0; col < indexMap.cols; ++col)
		{
			if ((int)*currentPixel++ == 255) continue; // ignore background

			result.ptr<Vec3b>(row)[col][1] = 120;
			result.ptr<Vec3b>(row)[col][0] = 120;
			result.ptr<Vec3b>(row)[col][2] = 120;
		}
	}

	return result;
}
```

yijiew, 2020-02-20:
Hi @eshirima , one possible issue in your code is that the color image is BGRA format, which has 4-bytes per pixel instead of 3. But you are casting the color image to result.ptr<Vec3b>. Could you try to change it to result.ptr<Vec4b> and see whether it solves the problem?

eshirima, 2020-02-20:
Sorry @yijiew but I do not understand what you mean. Did you mean to say change it to `colour.ptr`? Correct me if I am wrong but regardless of the colour orientation, shouldn't the body map still correctly map to the colour camera? Isn't the problem stemming from the transformation itself and not the pixel orientation? Especially at the end, it is vivid that the map is not properly _stretched_ out to fit the colour camera image as shown [here](https://docs.microsoft.com/en-us/azure/kinect-dk/use-image-transformation). I converted the colour image to OpenCV Mat as shown [here](https://stackoverflow.com/a/57222191/4962554) and simply changed the type to `CV_8UC1` for the body map in colour space since it returns a single channel image.

yijiew, 2020-02-20:
Oh. My bad... I just realized that I missed the most important part in my instruction :P Try to change it to  result.ptr<Vec4b>.

eshirima, 2020-02-20:
@yijiew your comment now makes more sense. And yes, that did solve the issue. I really appreciate it.

closed 2020-02-20 by eshirima