microsoft/Azure-Kinect-Sensor-SDK #1083: SDK documentation disappearing, no long-term device support

diablodale, 2020-02-18:
Kinect SDK documentation for v1 and v2 sensors is gone or disappearing. ⚠ Microsoft is not demonstrating support for the Kinect sensor to be used at mass-scale, factories, or other industries in which support for the devices is needed in timescale of years.

I request this Kinect team to address this issue by maintaining documentation to your sensor hardware. Either keep it live on a website...or publish the reference API documentation to a PDF file and publish that PDF. This is needed for the v1, v2, and Azure sensors.

I have for one year (since Feb 2019) escalated this to the MicrosoftDocs team. However, they have ignored contact, made no progress, and instead the documentation for Kinect sensors is further disappearing. https://github.com/MicrosoftDocs/feedback/issues/1214

The solutions that we build that include the Kinect are purchased and used by our customers for a period longer than 2-3 years. Our companies must maintain these solutions over the long-period. We must react to security issues, bugs, and regular updates. We need the reference documentation to do that.

tesych, 2020-02-18:
@diablodale, we reached out to the team responsible for the documentation and hopefully it will be available soon.

closed 2020-02-19 by tesych

diablodale, 2020-02-19:
Hi. You closed this without providing the resolution. What are the resolutions on this topic?

1. Have you confirmed that the documentation now exists?
2. Does this Kinect team commit to the sensor and the necessary references for coming years?

Please do not toss this issue to the support/doc team. They are not the product team and do not own priorities, headcount, and customers. Only the product team can drive auxiliary teams like support/doc to do work. Only the product team owns the success of the Kinect product.

Please do share the resolution for us all. Your customers want to know what the Kinect team's commitment is, so that we can also align our commitment.

tesych, 2020-02-27:
@diablodale I closed this issue because this issue already being tracked at [Microsoft documentation GitHub repo](https://github.com/MicrosoftDocs/feedback/issues/1214) and I connected the right people to work on it.

diablodale, 2020-02-28:
Below I outline gaps in your v3 Azure Kinect. Of course, this issue is a strategic one that uses v1/v2 as examples. So please broaden your view to the strategic issue here.

### 100% fail track-record

Microsoft has a 100% track-record of deleting Kinect reference documentation, it does not engender trust. The current Kinect Azure aka v3 references are insufficient.

Lifecycle https://support.microsoft.com/en-us/lifecycle/search?alpha=Azure%20Kinect%20SDK
leads to
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/microsoft-support.md
which in summary writes Kinect SDK v3 is only supported for **3 years**.

Therefore, your customers that build solutions have Microsoft support for 3 years and after that customers must self-support. Naturally, as that above link writes, we need to download and store material locally.

Great. Lets check that.

I see large gaps. I acknowledge it is possible that I have made mistakes below. If so, please do highlight my mistakes. It might also highlight a gap in Microsoft's public information that led to my inability to find such.

### Missing Kinect Azure SDK references

Your use of doxygen on the v3 Kinect SDK can allow customers to use locally kept GIT repos to recreate reference documentation for that SDK. 👍

What about all the references/docs that is not an API? The setup guides, etc.? How can customers download and keep local copies of setup guides, etc.? 🚨

### Missing Kinect Azure Body Tracking SDK references

Where is any of the the references/docs for the Body Tracking SDK?
https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download
leads to
https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/index.html
That is not downloadable, archive-able, not portable for customers to use locally.

No GIT repo available. No ability re-generate API references. 🚨

What about all the references/docs that is not an API? The setup guides, etc.? How can customers download and keep local copies of setup guides, etc.? 🚨

tesych, 2020-03-02:
@diablodale , Azure Kinect DK documentation is [GitHub open sourced  repository](https://github.com/MicrosoftDocs/azure-docs/tree/master/articles/kinect-dk), that you can create a local copy and work with it like any other GitHub repo.  Body tracking API documentation cannot be open sourced, but you can make a copy by using available tools.

diablodale, 2020-03-03:
🚨 Agreed, just as I wrote. There is no known way to make offline Body tracking SDK docs. 🚨
I have sold over 1000 Kinect solutions, with multiple Kinect within each solution. And sold across 51 countries so far.

Unfortunately, the link you provided has no known tools. You only provided a bing search query in hope something would show it. It didn't. No tools.

If Microsoft has a way for customers to make offline copies of the Body tracking SDK, please do share. If Microsoft doesn't, then I consider this a show stopper for Kinect v3.

No reference doc -> no solutions.

diablodale, 2020-03-05:
Thank you for our offline conversations to sort through everything. 👍 Anyone else that reads this issue can get specific answers and doc downloads at...
* Kinect v1/v2 all docs issue https://github.com/MicrosoftDocs/feedback/issues/1214
* Azure Kinect (aka Kinect v3) core guides/SDK docs in github repo listed above and use the 'Download PDF' link at the bottom left of https://docs.microsoft.com/en-us/azure/Kinect-dk/
* Azure Kinect (v3) Body Tracking SDK docs issue https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1100