microsoft/Azure-Kinect-Sensor-SDK #1086: One Azure Kinect Sensor not working

MasayaAUT, 2020-02-20:
Hi there,

I'm currently having issues with one of my Azure Kinect Sensors. I am currently using 2 azure kinects and a kinect v2 alongside the brekel software, everything was working fine at the start but now one of the Azure Kinect Sensors is not being picked up by the brekel software or the Viewer. The other Azure Kinect Sensor still works perfectly.

When refreshing the viewer this is the error that comes up:
[ error ] : find_libusb_device(). Unable to find Container ID: {27425d48-b481-4111-a593-a89e9f44d4ca}
[ error ] : usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()
[ error ] : colormcu_create(container_id, &device->colormcu) returned failure in k4a_device_open()

The sensor is being picked up by device manager and works in windows Camera software.
I have tried switching out cables and reinstalling the drivers but haven't been able to find a solution yet.

When checking the firmware on the sensor I also get this error:
[2020-02-21 11:21:49.076] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (364): find_libusb_device(). Unable to open LIBUSB at index 0
[2020-02-21 11:21:49.079] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
[2020-02-21 11:21:49.250] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (353): find_libusb_device(). Unable to find Container ID: {27425d48-b481-4111-a593-a89e9f44d4ca}
[2020-02-21 11:21:49.251] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (53): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()
[2020-02-21 11:21:49.256] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\firmware\firmware.c (164): colormcu_create(container_id, &firmware->colormcu) returned failure in firmware_create()
[2020-02-21 11:21:49.937] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (353): find_libusb_device(). Unable to find Container ID: {27425d48-b481-4111-a593-a89e9f44d4ca}
[2020-02-21 11:21:49.938] [error] [t=33960] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (53): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()

The issue only seems to be with this one computer I am currently on as well, as the sensor works on a different computer.

Anyone have any other troubleshooting steps I could try?

Thanks in advance.

wes-b, 2020-02-20:
The issue is probably related to the USB Host controller your USB 3.X port is connected to internally.  Using #384 as a reference, use Device Manager and view-> by connection to ensure depth / color camera's are connected to a host controller that is either Intel, TI, or Renesas, such as "Intel(R) USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)". There may be others that work, but these 3 we have tested. Also ensure the USB host controller has (Microsoft) on the end of the name to indicate the Microsoft driver is running rather than the manufacturers driver.

MasayaAUT, 2020-02-20:
Hi,

Thanks for your reply.
I had a look and it seems to be under a Intel USB 3.0 controller. I plugged both sensors in to see where the other one was going as well.

[image]

I just tried uninstalling the Azure Kinect Depth Camera to see if I could reinstall it, but now it seems to have completely disappeared.

[image]

Anyway to get that back?
[image]

MasayaAUT, 2020-02-20:
Thanks for your help!

Restarting my computer seemed to reinstall the Azure Kinect Depth Camera driver and everything is working again now!

closed 2020-02-20 by MasayaAUT

ScatterCory, 2023-01-19:
Just adding on to this to say that I have run into this problem as well, and it wasn't fixed until I uninstalled **ALL** of the Azure Kinect's component devices, then restarted my computer.
[image]