microsoft/Azure-Kinect-Sensor-SDK #1088: Crash when listing devices

tostercx, 2020-02-24:
**Describe the bug**
Running any of the prebuilt binaries or compiling the sample projects seem to crash when listing devices. I've tested it on 4 computers that I have access to - seems to crash on 2 of them :/ The computers have similar hardware tho, but not sure how to debug / what to look for.
Seems to have something to do with libusb? Or maybe with the USB drivers or the hardware itself? I've confirmed they are USB 3.0 ports and the device shows up correctly in device manager.

**To Reproduce**
1. run any binary from Azure Kinect SDK v1.3.0\tools
2. echo %errorlevel%
-1073741819 (that's 0xFFFFFFFFC0000005, so an access violation?)

**Expected behavior**
No crash.

**Logs**
I couldn't get the logger to output anything, I set the env vars that are supposed to enable logging but the best I got was `New logging session started` and only when not listing/doing anything with devices with the command line tools - either it crashes before it can write or I'm doing it wrong somehow.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10, 1909
 - SDK Version:  v1.3.0, also the same with lower versions
 - Firmware version:  ? factory default

**Additional context**
Stack trace looks similar with other command line tools and the kinect fusion example, not sure if that helps.

wes-b, 2020-02-24:
Is this an optimized build (like debug)? I am wondering if we can trust that callstack. In device manager   which your view to connection, are you connected to an Intel, TI, or Renasas USB 3.0 eXtensible Host Controller with the _(Microsoft)_ driver running on it?

tostercx, 2020-02-25:
I didn't build those myself, it's the executables that came with the SDK. I assume it's a release build with the pdb files kept?

I tried a VIA PCI USB card w/ MS drivers and built in Intel USB ports, also with MS drivers (lower in the picture, currently disconnected), same problem on both.

[image]

Yes, the actual problem for the crash is likely just before what happens on the call stack. Did you see the commit I linked https://github.com/microsoft/Azure-Kinect-Sensor-SDK/commit/413004e2c88d6f2c65f32a90ad11589eff779371 ? It's probably not in the latest release yet and if the device count is negative it proceeds with the data as if there are devices found.

wes-b, 2020-02-25:
Virtual USB controller is a new one. Good find.

closed 2020-02-25 by wes-b