microsoft/Azure-Kinect-Sensor-SDK #1089: Azure Kinect Viewer: failed to set media type at start

gladys0313, 2020-02-25:
Hello,

I started to try the Azure Kinect DK today. I followed the steps on [this webpage](https://docs.microsoft.com/en-gb/azure/Kinect-dk/set-up-azure-kinect-dk) closely. In the ka4viewer, when I started the device, it always gave me the following errors:
```
[ error    ] : Start(). Failed to set media type at start: 0x800706bf
[ error    ] : color->m_spCameraReader->Start(width, height, fps, config->color_format, &color_capture_available, color) returned failure in color_start()
[ error    ] : color_start(device->color, config) returned failure in k4a_device_start_cameras()
```
For your information, this is the complete log file:
```
[ info     ] : find_libusb_device(). Container ID found: {a5f491f6-09a5-4019-73a6-01e24ca5284c} 
[ info     ] : populate_serialnumber(). Serial Number found 000430794512
[ info     ] : find_libusb_device(). Container ID found: {a5f491f6-09a5-4019-73a6-01e24ca5284c} 
[ info     ] : populate_serialnumber(). Serial Number found 000430794512
[ info     ] : ******************** Device Info ********************
[ info     ] : K4A SDK version:     1.3.0
[ info     ] : Serial Number:       000430794512
[ info     ] : RGB Sensor Version:  1.6.102
[ info     ] : Depth Sensor Version:1.6.75
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
[ info     ] : ****************************************************
[ info     ] : k4a_device_start_cameras(). Starting camera's with the following config.
[ info     ] : k4a_device_start_cameras().     color_format:3
[ info     ] : k4a_device_start_cameras().     color_resolution:1
[ info     ] : k4a_device_start_cameras().     depth_mode:2
[ info     ] : k4a_device_start_cameras().     camera_fps:2
[ info     ] : k4a_device_start_cameras().     synchronized_images_only:1
[ info     ] : k4a_device_start_cameras().     depth_delay_off_color_usec:0
[ info     ] : k4a_device_start_cameras().     wired_sync_mode:0
[ info     ] : k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[ info     ] : k4a_device_start_cameras().     disable_streaming_indicator:0
[ info     ] : dynlib_create(). Depth Engine loaded C:\Program Files\Azure Kinect SDK v1.3.0\tools\depthengine_2_0.DLL
[ info     ] : verify_plugin(). Loaded Depth Engine version: 2.1.1
[ error    ] : Start(). Failed to set media type at start: 0x800706bf
[ error    ] : color->m_spCameraReader->Start(width, height, fps, config->color_format, &color_capture_available, color) returned failure in color_start()
[ error    ] : color_start(device->color, config) returned failure in k4a_device_start_cameras()
[ info     ] : k4a_device_start_cameras(). k4a_device_start_cameras started
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopping
[ info     ] : queue_disable(). Queue "dewrapper" waiting for blocking call to complete.
[ info     ] : queue_stop(). Queue "Queue_capture" stopped, shutting down and notifying consumers.
[ info     ] : queue_stop(). Queue "Queue_depth" stopped, shutting down and notifying consumers.
[ info     ] : queue_stop(). Queue "Queue_color" stopped, shutting down and notifying consumers.
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopped
```

I'm sure that:
1. the hardware is properly connected and charged; 
2. the firmware was updated successfully; 
3. the hardware (cameras, microphones) are found by the machine, as shown in the device manager; 
4. I have tried several usb 3.0 ports and I have tried two laptops; 
5. I'm using lenovo T460p and lenovo T470p. Both meet the minimal system requirements. 

From the viewer, the microphone seems to stream data normally, but three cameras and the IMU dont work. 

Does anyone have any idea about this?

wes-b, 2020-02-25:
The error code of 0x800706bf is key to this issue. See #853, #381, and #878.

wes-b, 2020-03-02:
Any update on this? I would love to hear about any potential fixes to resolve the issue

spigge, 2020-03-03:
I have the same problem. I can however run the depth camera in the viewer, it's only the RGB camera that refuses to work in my case.

wes-b, 2020-03-03:
Usually in these cases we find something is installed in the color camera pipeline and 3rd party software needs to be removed. I have seen certain gaming laptops install 3rd party apps (in the background) that breaks this pipeline. Meaning the machine runs well for a day after Windows install and then starts failing because it updated.

spigge, 2020-03-03:
Also for me, removing the Realtek folder as in #853 solved the problem.

closed 2020-03-13 by wes-b