microsoft/Azure-Kinect-Sensor-SDK #1090: Calculating XYZ depth for a point of interest (pixel) - different methods and different results

carlos-martinez, 2020-02-25:
Hi,

I want to get the XYZ depth values of something I found using the color camera. I read the documentation and guidelines, and still have some questions of the right way of doing this. There might be several questions on this post, so please let me know if it is better to split this in multiple posts.

- Scenario: We found a point of interest in the RGB image (e.g. ArUco marker)
- Goal: Get the XYZ depth of that point of interest

The _k4a_calibration_2d_to_3d_ method seems the be the right method to use, we need to provide the pixel coordinate and the depth value. Based on the documentation, we can use _k4a_transformation_depth_image_to_color_camera_ to calculate the depth. This method is used by the transformation_example.

The transformation_example provides guidance of two ways to combine depth and color: (1) transforming depth into color and (2) transforming color into depth. It seems that I could use the colored point cloud in order the get the XYZ depth of my point of interest (since this is kind of what we need) . Interestingly, the calculated point clouds (from the two methods) were shifted one from the other.

So this leads me to the following options for my goal:

A.	Use color -> depth method from transformation_example
B.	Use depth -> color method from transformation_example
C.	Use the depth of (A) in the method  k4a_calibration_2d_to_3d
D.	Use the depth of (B) in the method  k4a_calibration_2d_to_3d

Pseudo-code

```
config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
config.color_resolution = K4A_COLOR_RESOLUTION_720P;
config.synchronized_images_only = true;
config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
config.camera_fps = K4A_FRAMES_PER_SECOND_15;
int16_t pointX, pointY, pointZ;
int16_t pointX_rev, pointY_rev, pointZ_rev;
string filename;
string filename2;
string filename_rev;
string filename2_rev;

// Option A - this method is based on the transformation_example, but just returns the data we are interested on
point_cloud_color_to_depth(transformation, depth_image, color_image,
	filename,
	nearestRow, nearestCol,
	&pointX, &pointY, &pointZ);
// Option B -  this method is based on the transformation_example, but just returns the data we are interested on
point_cloud_depth_to_color(transformation, depth_image, color_image,
	filename_rev,
	nearestRow, nearestCol,
	&pointX_rev, &pointY_rev, &pointZ_rev);

std::cout << "A - Transformed 3D (via color -> depth) - XYZ: " << pointX << " " << pointY << " " << pointZ << "\n";
std::cout << "B - Transformed 3D (via depth -> color) - XYZ: " << pointX_rev << " " << pointY_rev << " " << pointZ_rev << "\n";

if (K4A_RESULT_SUCCEEDED == k4a_calibration_2d_to_3d(&calibration, &p, pointZ, K4A_CALIBRATION_TYPE_COLOR, K4A_CALIBRATION_TYPE_DEPTH, &ray, &valid))
{
	if (valid)
	{
		std::cout << "C - Calibration 2d_to_3d (w / Option A) - XYZ: " << ray.xyz.x << " " << ray.xyz.y << " " << ray.xyz.z << "\n";
	}
}
else
{
	std::cout << "    Conversion to 3D FAILED!\n";
}

if (K4A_RESULT_SUCCEEDED == k4a_calibration_2d_to_3d(&calibration, &p, pointZ_rev, K4A_CALIBRATION_TYPE_COLOR, K4A_CALIBRATION_TYPE_DEPTH, &ray, &valid))
{
	if (valid)
	{
		std::cout << "D - Calibration 2d_to_3d (w / Option B) - XYZ: " << ray.xyz.x << " " << ray.xyz.y << " " << ray.xyz.z << "\n";
	}
}
else
{
	std::cout << "    Conversion to 3D FAILED!\n";
}

```

Output:

```
Oringal 2D - XY: 585 425.5
A - Transformed 3D (via color -> depth) - XYZ: -26 9 427
B - Transformed 3D (via depth -> color) - XYZ: -38 42 443
C - Calibration 2d_to_3d (w / Option A) - XYZ: -3.87379 -0.7335 425.217
D - Calibration 2d_to_3d (w / Option B) - XYZ: -5.21588 -0.847239 441.292
```

Each option provides different answer. 

I would appreciate any insights or help to understand the right option, and what are the relationships or trade-offs among these methods.  Part of that includes why A and B have a shift between their outputs (up to several mm when items are within 400 mm of the camera, up to several cm at longer distances).

Thanks

rabbitdaxi, 2020-03-04:
Great question! Your evaluation is end to end check involving many steps and the difference may coming from few factors. In general, you can always refer to the reference code of depth to color and color to depth transform which is open sources here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/rgbz.c. You will learn from the code that given the fact the image resolution of depth and color are different, there are interpolation involved in both direction. For both direction transformation, they consumer color and depth intrinsics and extrinsics. And during the transformation of depth to color, you will also consume depth value, which can contain certain systematic and random error. You can refer to #1058 which contains some   description of how the transformation works

carlos-martinez, 2020-03-05:
Thanks for your feedback.

I found these links before, but I re-read them and have a little more understanding of the differences. Based on issue https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1058, it is not possible get a relationship between two methods. Is this correct, or is there a proposed method for this?

It seems that the "depth_to_color" provides a better alignment with the world. The challenge is that I am trying to match the pixel found within a point cloud generated using the openCV kinfu. The Azure Sample Kinfu implementation uses the distortion method to gather the depth_image which is then fed to the Kinfu engine.. This seems to be more aligned with the "color-to-depth". This is why I need this transformation.

Hope the question still make sense.

Thanks in advanced.

rabbitdaxi, 2020-07-17:
@carlos-martinez I haven't been keep up with GitHub for a while, and I re-read your issue. Actually, I think my previous comment was not really answering your questions even though it pointed to you the right documentation to look at. 

I have a few comments from your original post and hope it helps:
- I cannot see your full code, but if you are based off the transformation example, please also pay attention to that the color point cloud from point_cloud_color_to_depth and point_cloud_depth_to_color are in difference camera space (former is in [depth camera space](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61951daac782234f4f28322c0904ba1c4702d0ba/examples/transformation/main.cpp#L48), latter is in [color camera space](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61951daac782234f4f28322c0904ba1c4702d0ba/examples/transformation/main.cpp#L102)). That could partially explain your value output are not the same for the first two printf rows (A and B). However, I am not sure how the "nearestRow, nearestCol" is inside your implementation, but you should debug a little bit from this perspective. Ideally, you should compare the same point in the same camera space. 
- Same for the last two (C and D), you are assuming both point cloud are in the color camera space when you call the 2d to 3d unprojection, that is true for point_cloud_depth_to_color, but not for point_cloud_color_to_depth

In short, for your original purpose, the recommended thing to do is:
- you transform depth image to the color image space using k4a_transformation_depth_image_to_color_camera
- then you will get a depth image has 1:1 pixel correspondence with the color image
- then for a given RGB pixel (i, j) that you are interested, you can read the depth value from the transformed depth image with the pixel index (i, j) -> d, then you can call k4a_calibration_2d_to_3d with i, j and d as input to get the 3d XYZ.