microsoft/Azure-Kinect-Sensor-SDK #1091: k4abt_simple_3d_viewer crahs

Zafoue, 2020-02-26:
**Describe the bug**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
-Body track version 1.00
 - Firmware version:  < !AzureKinectDK_Fw_1.6.102075014 -->

wes-b, 2020-02-26:
Sorry we need more complete information about the bug.

Zafoue, 2020-02-26:
> Sorry we need more complete information about the bug.

When I launch the body tracker after 2 minutes the application crashes and stops working. We want to confirm that this is not a driver problem I tested the body tracker in cpu mode and the no problem
 - OS with Version: <! win 10 v 1809 -->
 - SDK Version:  <! V 1.3 -->
-Body track version 1.00
 - Firmware version:  < !AzureKinectDK_Fw_1.6.102075014 -->

wes-b, 2020-02-26:
Our bug template covers these questions too, can you provide
* steps to reproduce the problem
* logs captured while running - https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs

Zafoue, 2020-02-27:
> Our bug template covers these questions too, can you provide
> 
>     * steps to reproduce the problem
> 
>     * logs captured while running - https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#collecting-logs

steps to reproduce the problem
lunch body track
wait 2 mn 
crash 
I have created a dmp file to you 
[Editor.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4260964/Editor.log)
[error.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4260968/error.log)
[crash.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4260970/crash.txt)
just modify the extension of crash.txt to crash.dmp it dmp file

AswinkarthikeyenAK, 2020-02-29:
Hi, 

I am having trouble launching the k4abt_simple_3d_viewer CPU which throws an error,

Shader Error: 0:1(10): error: GLSL 4.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

Any ideas on how to solve this problem? 
I am using Ubuntu 18.04.

rfilkov, 2020-03-03:
Hi @yijiew , is this the same issue with the onnxruntime as before?

AswinkarthikeyenAK, 2020-03-04:
@rfilkov, Thanks. The issue is solved now. I had problems because ubuntu stopped my Nvidia graphic card to load during boot.

rfilkov, 2020-03-04:
Thank you for the info, @AswinkarthikeyenAK . My question to @yijiew was in regard to the original issue, reported by @Zafoue .

yijiew, 2020-03-09:
Hi @Zafoue and @rfilkov, sorry for responding to this issue late. Everything slowed down a bit last week given the coronavirus situation in Seattle area. Hope everything is fine in your countries. 

Thanks for creating the crash dump. I was able to open the dump and located the crash. It actually crashed inside our processing code instead of the onnxruntime loading. But it crashed in a place that we cannot easily reason the cause of the bug without a repro locally. 

@Zafoue, could you provide the following information to help us debug the issue? 
1. Did you hit the crash very time you launch the body tracking for 2 minutes? Or it is a crash you only see occasionally? 
2. Do you find a pattern of the crash? 
3. What GPU are you using? 
On our side, we have a test machine running 24/7 for almost 2 months and we didn't see a single crash...

qm13, 2020-03-16:
@Zafoue and @rfilkov can you make recordings that repro the issue? If so we can arrange for you to upload the recording to Microsoft.

Zafoue, 2020-03-16:
Hi, sorry for the late,

1)Yes it when i use body track viwer for 2 mn or less.
2) not yet
3)gtx 980 ti

Zafoue, 2020-03-17:
> Hi @Zafoue and @rfilkov, sorry for responding to this issue late. Everything slowed down a bit last week given the coronavirus situation in Seattle area. Hope everything is fine in your countries. [...]

 Hi, sorry for the late,

1)Yes it when i use body track viwer for 2 mn or less.
2) not yet
3)gtx 980 ti

Zafoue, 2020-03-17:
I have tested Azure Kinect Body Tracking SDK 1.0.1 and i have already the bug :/

closed 2020-04-14 by qm13