microsoft/Azure-Kinect-Sensor-SDK #1093: Publish SDK 1.4.0

tesych, 2020-02-26:
alpha 1.4.0 version is released and include support for ARM.

tkircher, 2020-02-26:
Where can we find the depth library for ARM?

tesych, 2020-02-27:
@tkircher all the information is coming in the next few days. I know everyone is very excited about this release :)

tkircher, 2020-02-27:
Definitely excited. Got a Jetson Nano specifically for this.

airfield20, 2020-02-28:
Following for arm information

ergodaveh, 2020-02-28:
Where would I find a Windows Installer for this release?

wes-b, 2020-02-28:
This issue is still a TODO, title updated to reflect that.

wes-b, 2020-03-03:
Updated description and added Nuget Link

UnaNancyOwen, 2020-03-04:
How is ARM64 support on Azure Kinect Body Tracking SDK going?

tesych, 2020-03-04:
@UnaNancyOwen , ARM support for Body Tracking SDK is in progress and coming soon. I will update with more details.

tkircher, 2020-03-05:
Incidentally, after upgrading to AzureKinectDK_Fw_1.6.107078014.bin, from the nuget 1.4.0 release, the Kinect stopped responding. Factory reset did not work, and the LED is blinking white.

tkircher, 2020-03-05:
Incidentally, confirmed that k4aviewer works on the Jetson Nano (arm64).

wes-b, 2020-03-05:
Description updated, [1.4.0-alpha.3](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.3) replaced with 
[1.4.0-alpha.4](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4)

UnaNancyOwen, 2020-03-06:
Please consider merge these pull requests (#822, #1097) before to release v1.4.0.
It provide all SDK APIs in C# wrapper by adding missing functions.

SladeLu, 2020-03-07:
> Where can we find the depth library for ARM?

same question

tkircher, 2020-03-07:
The ARM depth library is in the NuGet package, under linux/lib/native/arm64/

The link is here:

https://www.nuget.org/api/v2/package/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4

and the file is actually just a zip file. So rename it from .nuget to .zip and unzip it.

SladeLu, 2020-03-07:
> The ARM depth library is in the NuGet package, under linux/lib/native/arm64/
> 
> The link is here:
> 
> https://www.nuget.org/api/v2/package/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4
> 
> and the file is actually just a zip file. So rename it from .nuget to .zip and unzip it.

Thanks, I know where to find the libdepthengine2.0 file now, but still don't know where to place it under arm64 architecture to run k4aviewer sucessfully.

tkircher, 2020-03-07:
You almost certainly want to put it in /lib/aarch64-linux-gnu/

valdivj, 2020-03-07:
How did you get 1.4.0-alpha.4 to install on jetson nano.
I have nuget installed put it doesn't want to pull the package?
Any ETA on the .deb package

nano@nano-desktop:~$ nuget
NuGet Version: 2.8.7.0

SladeLu, 2020-03-07:
> You almost certainly want to put it in /lib/aarch64-linux-gnu/

THANKS! My jetson nano works well!

valdivj, 2020-03-07:
got it had to do this:
alias nuget="mono /usr/local/bin/nuget.exe"

valdivj, 2020-03-08:
Got the Kinect for Azure running on the Jetson Nano. :)
Anybody have any ideas on making the Kinect for Azure run with Gstreamer.
I want to use the Kinect for Azure with Deepstream.

I have already done this with the Kinect V2
https://youtu.be/MiVizFsyMRU 

I know the Kinect V2 & the Kinect for Azure are different but it shouldnt be impossible.

valdivj, 2020-03-09:
If anybody is interested here is a video of the Kinect for Azure running on the Jetson Nano
https://youtu.be/NpavmEy9zJU

eric-schleicher, 2020-03-09:
@valdivj what is the (full) command line argument you used.   I'm new to nuget as well.  the syntax isn't intuitive.  also did you have to update the firmware on the K4A?

valdivj, 2020-03-10:
Had to install mono first
Then ran this:$ alias nuget="mono /usr/local/bin/nuget.exe"
Then this:$ nuget install Microsoft.Azure.Kinect.Sensor -Version 1.4.0-alpha.4
You might want to try this :https://github.com/microsoft/Azure-Kinect-Sensor-SDK
They just added arm support 3 days ago.

iscsrl, 2020-03-10:
Is it possible to use it on RaspberryPi 4? Any tips on how can I perform the installation?

tkircher, 2020-03-10:
> Is it possible to use it on RaspberryPi 4? Any tips on how can I perform the installation?

The short answer is no.

SladeLu, 2020-03-11:
BTW，does anyone try applying Azure SDK ROS package on JETSON Nano? I failed during compling the ROS azure package. Maybe need to modify cmakelist.txt to adapt the arm64 version sdk?

wes-b, 2020-03-11:
Beta Released today

tesych, 2020-03-11:
@UnaNancyOwen we will include #1097 but not #822 since we need more time to review.

UnaNancyOwen, 2020-03-11:
@tesych Thanks! Why is #822 stagnant? It seems that reviews have been paused for several months.

tesych, 2020-03-12:
@UnaNancyOwen we run into few technical issues, that we are working through and will release as soon as possible.

UnaNancyOwen, 2020-03-12:
@tesych I see. I'm looking forward to provide it in v1.4.x (or v1.5.0). Thanks, :)

valdivj, 2020-03-12:
Here is a link to github repo with instructions to get Azure for Kinect to run on nano
https://github.com/valdivj/Azure-for-Kinect-Jetson-nano

eric-schleicher, 2020-03-16:
As @valdivj mentions, this is the biggest event since the release of the camera.  many thanks!

nonlinear1, 2020-03-20:
Good news, But I just have Nvidia Xavier, Could the SDK with ARM support can be applicable to Nvidia Xavier? if not, could you give the Nvidia Xavier arm support.

wes-b, 2020-03-20:
> Good news, But I just have Nvidia Xavier, Could the SDK with ARM support can be applicable to Nvidia Xavier? if not, could you give the Nvidia Xavier arm support?

Sorry @nonlinear, we don't have any Xavier's to test against. I am thinking it should work though.

richardeakin, 2020-03-20:
> @UnaNancyOwen , ARM support for Body Tracking SDK is in progress and coming soon. I will update with more details.

Any updates on this? We'd like to test on an NVidia Jetson TX2.

eric-schleicher, 2020-03-20:
@nonlinear1 it absolutely works on the jetson nano, so you really shouldn't any issue on the xavier

valdivj, 2020-03-22:
This a little off topic but if  you are going to use the Jetson Nano and the Kinect for Azure here is a link to a video that shows you how you can easily make a portable Battery power supply to run the pair.
https://youtu.be/WuC0PN5PuYA

SladeLu, 2020-03-23:
Does someone build this new SDK and successfully complie AZURE_ROS_PKG on arm64(jetson nano). Wondering some instructions using this ARM64 SDK on AZURE_ROS_PKG.  Seems the manully compling file structure cannot fit the AZURE_ROS_PKG required provided by microsoft.  Or it will have some updates on AZURE_ROS_PKG ?

nonlinear1, 2020-03-23:
Just now, I try to compile the Azure Kinect SDK for ARM in Xavier. But I failed. I first download  1.4.0-alpha.4 from https://www.nuget.org/api/v2/package/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4. Then I copy the /linux/lib/native/arm64/release/libdepthengine.so.2.0 from 1.4.0-alpha.4 to "usr/lib" in Xavier. After that, I download the Azure Kinect SDK from https://github.com/microsoft/Azure-Kinect-Sensor-SDK. Is something wrong? Could someone know? Thanks a lot. the following is  a snapshot of error.
[image]

wes-b, 2020-03-23:
@nonlinear1, you are close. [This](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/scripts/docker/setup-ubuntu.sh) docker file script manages the dependencies that ARM64 builds need. You should be able to use this to get your dependencies installed. This script is for cross compiling ARM64 on AMD64, so don't run the script directly on ARM64 or you will end up with AMD64 binaries.

nonlinear1, 2020-03-24:
Thank you webs-b I have succeed in compiling the Azure Kinect SDK in Xavier. OS: ubuntu 18.04. This is the snap shot:
[image]

TheRunningHorse, 2020-03-24:
I tried to compling the SDK in Xavier too, but I failed.Is there are something wrong?

FAILED: tests/Transformation/CMakeFiles/transformation_ut.dir/transformation.cpp.o 
/usr/bin/c++   -I../extern/azure_c_shared/src/inc -I../extern/azure_c_shared/src/pal/linux -isystem ../extern/googletest/src/googletest/include -isystem ../extern/googletest/src/googletest -I../src/../include -Isrc/sdk/include -I../extern/spdlog/src/include -I../tests/UnitTests/utcommon/inc -isystem ../extern/googletest/src/googlemock/include -isystem ../extern/googletest/src/googlemock -I../include -O2 -g -DNDEBUG -fPIE -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -Wno-missing-field-initializers -Werror -pthread -std=gnu++11 -MD -MT tests/Transformation/CMakeFiles/transformation_ut.dir/transformation.cpp.o -MF tests/Transformation/CMakeFiles/transformation_ut.dir/transformation.cpp.o.d -o tests/Transformation/CMakeFiles/transformation_ut.dir/transformation.cpp.o -c ../tests/Transformation/transformation.cpp
../tests/Transformation/transformation.cpp: In member function ‘virtual void transformation_ut_transformation_depth_image_to_point_cloud_Test::TestBody()’:
../tests/Transformation/transformation.cpp:403:43: error: call of overloaded ‘abs(double)’ is ambiguous
     if (std::abs(check_sum - reference_val) > 0.001)
                                           ^
In file included from /usr/include/c++/5/cstdlib:72:0,
                 from /usr/include/c++/5/ext/string_conversions.h:41,
                 from /usr/include/c++/5/bits/basic_string.h:5352,
                 from /usr/include/c++/5/string:52,
                 from /usr/include/c++/5/bits/locale_classes.h:40,
                 from /usr/include/c++/5/bits/ios_base.h:41,
                 from /usr/include/c++/5/ios:42,
                 from /usr/include/c++/5/ostream:38,
                 from ../extern/googletest/src/googletest/include/gtest/gtest.h:56,
                 from ../tests/UnitTests/utcommon/inc/utcommon.h:7,
                 from ../tests/Transformation/transformation.cpp:4:
/usr/include/stdlib.h:837:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^
In file included from /usr/include/c++/5/ext/string_conversions.h:41:0,
                 from /usr/include/c++/5/bits/basic_string.h:5352,
                 from /usr/include/c++/5/string:52,
                 from /usr/include/c++/5/bits/locale_classes.h:40,
                 from /usr/include/c++/5/bits/ios_base.h:41,
                 from /usr/include/c++/5/ios:42,
                 from /usr/include/c++/5/ostream:38,
                 from ../extern/googletest/src/googletest/include/gtest/gtest.h:56,
                 from ../tests/UnitTests/utcommon/inc/utcommon.h:7,
                 from ../tests/Transformation/transformation.cpp:4:
/usr/include/c++/5/cstdlib:166:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^
/usr/include/c++/5/cstdlib:174:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^
/usr/include/c++/5/cstdlib:179:3: note: candidate: __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^

UnaNancyOwen, 2020-03-24:
Azure Kinect worked on Jetson Nano! 👍 
I'm look forward to more optimization in the future.
[image]

closed 2020-03-24 by wes-b

UnaNancyOwen, 2020-03-25:
Has the Azure Kinect Body Tracking SDK with ARM64 supported been released?
Do we still have to wait a while for that? Thanks,

TheRunningHorse, 2020-03-25:
I have another error now!
FAILED: tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4apointcloudwindow.cpp.o 
/usr/bin/g++   -I../tools/k4aviewer -Isrc/sdk/include -I../include -Isrc/record/sdk/include -I../extern/imgui/src -I../extern/imgui/src/examples -I../extern/imgui/src/examples/libs/gl3w -I../extern/imgui/src/examples/libs/glfw/include -I../extern/libjpeg-turbo/src -I../extern/libusb/src/libusb -I../extern/libyuv/src -I../extern/libyuv/src/include -I../extern/glfw/src/include -O2 -g -DNDEBUG -fPIE -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -Wno-missing-field-initializers -Werror -idirafter/usr/include/soundio -std=gnu++11 -MD -MT tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4apointcloudwindow.cpp.o -MF tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4apointcloudwindow.cpp.o.d -o tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4apointcloudwindow.cpp.o -c ../tools/k4aviewer/k4apointcloudwindow.cpp
../tools/k4aviewer/k4apointcloudwindow.cpp: In member function ‘virtual void k4aviewer::K4APointCloudWindow::Show(k4aviewer::K4AWindowPlacementInfo)’:
../tools/k4aviewer/k4apointcloudwindow.cpp:95:12: error: ‘SeparatorEx’ is not a member of ‘ImGui’
     ImGui::SeparatorEx(ImGuiSeparatorFlags_Vertical);
            ^~~~~~~~~~~
../tools/k4aviewer/k4apointcloudwindow.cpp:95:12: note: suggested alternative: ‘Separator’
     ImGui::SeparatorEx(ImGuiSeparatorFlags_Vertical);
            ^~~~~~~~~~~
            Separator
[13/21] Building CXX object tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4awindowset.cpp.o

wes-b, 2020-03-25:
> I have another error now!
> FAILED: tools/k4aviewer/CMakeFiles/k4aviewer.dir/k4apointcloudwindow.cpp.o
> /usr/bin/g++ -I../tools/k4aviewer -Isrc/sdk/include -I../include -Isrc/record/sdk/include [...]

Please open a new bug and share steps to reproduce how you got to this build error. My guess is that you are working from the ZIP file download, which we do not support. You need to clode the repo so that submodules get populated and built.

UnaNancyOwen, 2020-04-02:
@tesych @wes-b Please update download link in Microsoft Docs. It's still v1.3.0.
https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download#windows-download-link

wes-b, 2020-04-02:
> @tesych @wes-b Please update download link in Microsoft Docs. It's still v1.3.0.
> https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download#windows-download-link

Thank you! I have submitted a PR.

valdivj, 2020-04-05:
If anybody is interested here is a video of the Kinect for Azure  running Nvida Deepstream using the Jetson Nano.
https://youtu.be/Pew5plgRKps

tkircher, 2020-04-06:
@valdivj using the drill body and battery is a great idea. This is mine:

[image]
[image]

valdivj, 2020-04-06:
@tkircher  Looks Clean!
All it needs is a label with your Brand on it. : )

dadaofeng, 2020-04-09:
This version (1.4.0) does not support body tracking sdk on Ubuntu 18.04.
Is there any way to deal with this?

wes-b, 2020-04-09:
> This version (1.4.0) does not support body tracking sdk on Ubuntu 18.04.
> Is there any way to deal with this?

Body Tracking team is also adding ARM support. A new version of BT is comming that will work with 1.4.0.

vnt1537, 2020-04-27:
> Thank you webs-b. I have succeeded in compiling the Azure Kinect SDK in Xavier. OS: ubuntu 18.04. This is the snap shot:
> [image]

Hello, can you send some link to some instruction where I can install this successfully for Nvidia Xavier!! It would be really helpfully :) Thank you.

wes-b, 2020-04-27:
>Hello, can you send some link to some instruction where I can install this successfully for Nvidia Xavier!! It would be really helpfully :) Thank you.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#debian-package

wes-b, 2020-04-27:
Locking thread to encourage the creation of new issues

PR #1149 (2020-03-24): Release SDK 1.4.0
## Fixes #1093

### Description of the changes:
- See Changelog.md