microsoft/Azure-Kinect-Sensor-SDK #1094: RGB frame is not sync in subordinate mode

etiennedub, 2020-02-27:
**Describe the bug**
When the Kinect is set in subordinate sync mode the RGB frame is out of sync with the input signal. 

I test the timming by producing a signal with an Arduino and turning off a led at the right time.
I observed that the first picture is taken after the second pulse with the firmware 1.6.102 and the 3th pulse with 1.6.104. That is not a problem but if you could confirm it, I would appreciate because I need to find the precise timming of the first frame relatively to the pulse.
In my test, I turn off a led ~22ms after the second pulse to see it in the picture. In the case where the led is turn off before, the led is off in the picture. In theory, I could turn off the led 500us (exposure time) after the rising edge.
This is the arduino output taken with a logic analyser:
[image]
The top signal is the sync signal at 30hz and the other one is the led. With a time of ~22ms between A1 and A2, I can see the led:
[image]
And with ~20ms the led is off:
[image]

**To Reproduce**

1. Set the kinect in subordinate mode with the minimal exposure (500 us) and `subordinate_delay_off_master_usec` set to 0.
2. Send two pulses and turn off the led 500us after the second rising edge.

**Expected behavior**
The led should be on on the rgb picture.

**Desktop (please complete the following information):**
 - OS with Version: Arch Linux x86_64 Linux 5.4.15-arch1-1 
 - SDK Version:  1.2.0 
 - Firmware version:  1.6.102 and 1.6.104

tesych, 2020-03-02:
@etiennedub thank you for the question. Please use the latest version of the SDK 1.3 and update to the latest available firmware and try repro the issue.

etiennedub, 2020-03-10:
I think it's not a bug but the problem is due to the rolling shutter.

closed 2020-03-10 by etiennedub