microsoft/Azure-Kinect-Sensor-SDK #1095: Tracker doesn't work in a WPF in published form

linatpensyl, 2020-02-28:
**Describe the bug**

Tracker.Create(this.kinect.GetCalibration(), new TrackerConfiguration()) works in debug and released WPF form but crashed in published form. 

**To Reproduce**
1. Clone this repository: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/build2019/csharp/1%20-%20AcquiringImages and open it in Visual Sutdio.

2. In mainwindow.xaml.cs, add this code:

```
 try
            {
                Tracker tracker = Tracker.Create(this.kinect.GetCalibration(), new TrackerConfiguration()
                {
                    ProcessingMode = TrackerProcessingMode.Gpu,
                    SensorOrientation = SensorOrientation.Default
                });
            }
            catch (Exception exception)
            {
                MessageBox.Show(exception.ToString());
            }
```
under `this.colorHeight = this.kinect.GetCalibration().ColorCameraCalibration.ResolutionHeight;` (around line 79). 
3. Run it in Debug or Release form, no error pop up, the program run as usual.

4. Click "Build" at the bar on the top of UI, click "Publish 1-AcuireImages", 
[image]

specify the folder you want to deploy this app to, then click "Next".
5. Choose "From CD-ROM or DVD-ROM", then click "Next"
6. Choose "The application will not check for updates", then "Next"
7. Click "Finish"
8. Wait for VS2019 to build the applcation then the folder will pop up, click 1-AcquiringImages.application

9. Error happen when creating tracker:
[image]

**Expected behavior**
It shouldn't have any error on this since we can run in Debug and Release mode.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 18363.657
 - SDK Version:  Tracker: 1.0.0 Sensor: 1.3.0
 - Firmware version:  

```
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**

yijiew, 2020-03-02:
Could you create a K4ABT log file following this instruction: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting?

linatpensyl, 2020-03-02:
Can I also create log even though in c#? I think the problems appears in k4abt.dll instead of k4a.dll. I also followed the instruction on the website but I didn't see the log file in my directory.

linatpensyl, 2020-03-02:
Hmm.. 
I am not able to collect the log file from k4abt.dll

Here is what I did on the command terminal:
1. set K4ABT_ENABLE_LOG_TO_A_FILE=C:\Users\AdobeTest\Desktop\k4abt.log
2. set K4ABT_LOG_LEVEL=DEFAULT
3. run 1-AcquiringImages.application

Error message pop up, but no log file on the desktop.

yijiew, 2020-03-18:
When you run the body tracking viewer that shipped with the SDK MSI installer, do you see any log files?

closed 2020-04-14 by qm13