microsoft/Azure-Kinect-Sensor-SDK #1096: k4a_playback_set_color_conversion is mandatory to use color images

emepetres, 2020-03-02:
**Describe the bug**
On v1.3.0, When working with a mkv recording, `k4a_playback_set_color_conversion` **must** be called in order to `k4a_playback_get_next_capture` return a capture with color information. However, documentation says that by default, if the function is not called, same format as recording will be used.

**To Reproduce**
Using v1.3.0 SDK, open a mkv recording with color information as you usually do. Do not set the color conversion. Then, get the capture and the color image. It will be null:
```
k4a_playback_open(....)
...
for
  k4a_playback_get_next_capture(...)
  k4a_capture_get_color_image(..) // <- **returns null!**
...
````

**Expected behavior**
If the video was recorded with color information, `k4a_capture_get_color_image` should not return null.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Pro x64
 - SDK Version:  1.3.0
 - Firmware version: 
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

wes-b, 2020-03-13:
I can't reproduce this. We have a test [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/52cfb4a87639b3db9cb176b8b9ac814e2b5a4900/tests/RecordTests/UnitTest/playback_ut.cpp#L69) that tests this. I even removed the call to k4a_playback_get_record_configuration and reran, no luck.

What are the settings used to record the MKV file?

emepetres, 2020-03-16:
Hi @wes-b -s. So I'm using now a clean PC with same set up, same behaviour. MKV is recorded with default values (attached in link below).

I made a small cmake project to reproduce the bug: If I use the capture from k4abt, and do `k4abt_playback_set_color_conversion`, color image is correct as I explain in the issue (but not when I don't call it). However, to make it simpler, I removed k4abt logic from it, and I'm not able to get the color image, using a real device or a recording, even calling `k4abt_playback_set_color_conversion`.

I'm sharing this simple project with you, so you can test it and check if I'm maybe doing something wrong, or there is a bug. The recording is inside the project:

https://we.tl/t-vheQrPJArL

The example uses opencv to render images.

closed 2021-08-02 by emepetres