microsoft/Azure-Kinect-Sensor-SDK #1100: Body Tracking API documentations need offline download

ETowleMS, 2020-03-04:
Current Body Tracking API documentation in Doxygen has no elligant solution for downloading the entire API set as a PDF (or other) for offline saving.

qm13, 2021-03-04:
Won't fix

closed 2021-03-04 by qm13

diablodale, 2021-03-04:
For those needing archive for long-term support, I was able to get a scrape of the most important body tracking APIs using the following. It is not pretty. Yet, start browser navigation at modules.html and the right pane content and links seem to work. 🤦‍♀️

```
wget --recursive --convert-links --page-requisites --restrict-file-names=windows --timestamping --level=inf --no-remove-listing --domains microsoft.github.io --no-parent --adjust-extension --tries=5 --wait=1 --random-wait https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/modules.html
```