microsoft/Azure-Kinect-Sensor-SDK #1101: Green Screen Error:

SaraHackett13, 2020-03-04:
I am able to compile and run the green_screen example. I am wanting to calibrate two cameras. I used the provided checkerboard image and made sure it was visible to both cameras. I am using the following arguments: green_screen 2 9 6 22

I get an error that says "ERROR: Timedout waiting for synchronized captures"

How can I fix this? 

Here are screenshots of my command line:
[image]
[image]

wes-b, 2020-03-04:
What firmware version are you using?

SaraHackett13, 2020-03-04:
@wes-b I am using Azure Kinect SDk v1.3.0

wes-b, 2020-03-04:
That is the SDK version, I believe you need to update the firmware on each device. 

```AzureKinectFirmwareTool.exe -q``` will report the version.

Review #530 for the issues with given firmware.

SaraHackett13, 2020-03-05:
I ran `AzureKinectFirmwareTool.exe -q` and got the following output for all three of my cameras. 

[image]

SaraHackett13, 2020-03-06:
I tried to update my firmware using 
```
AzureKinectFirmwareTool.exe -u firmware\AzureKinectDK_Fw_1.6.102075014.bin
```

The output was `SUCCESS: The firmware was already up-to-date.`

[This page](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md) shows that firmware 1.6.102075014 is the latest. 

How do I install firmware version 1.6.104075014?

wes-b, 2020-03-09:
Download this Alpha package [1.4.0-alpha.4](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.4) and use the firmware there, we are about to promote it to beta quality.

SaraHackett13, 2020-03-12:
I was able to get the Green Screen example working. 
Initially, I had three cameras powered on, connected to the host PC, and synced with 3.5mm cables. I disconnected the third camera. Then ran `green_screen 2 9 6 22`  again with success. I did _not_ need to update my firmware. 

Thanks for the help.

wes-b, 2020-03-12:
Good new, thanks for the update.

closed 2020-03-12 by wes-b