microsoft/Azure-Kinect-Sensor-SDK #1102: Help with undistorting color/IR images

SeanStephensenAIRM, 2020-03-04:
My goal is to have the color, depth and IR images all undistorted in color space. I have adapted the undistortion example and seem to be getting successful depth images (the "undistorted" file looks similar to the non-undistorted one, I haven't compared further), but the color image has major error, and the IR image fails to remap. Please find my code below, as well as the output "undistorted" color image. 

I think the source of the errors may relate to the stride bytes when I use k4a_image_create for the undistorted images, I still don't seem to have a good understanding of what should go there for each image type.

`pinhole = create_pinhole_from_xy_range(&cameras[ii].calibration, K4A_CALIBRATION_TYPE_COLOR);
k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(coordinate_t), &depth_lut);
k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(coordinate_t), &color_lut);
k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(coordinate_t), &IR_lut);

create_undistortion_lut(&cameras[ii].calibration, K4A_CALIBRATION_TYPE_COLOR, &pinhole, 								depth_lut, depth_interpolation);
create_undistortion_lut(&cameras[ii].calibration, K4A_CALIBRATION_TYPE_COLOR, &pinhole, color_lut, color_interpolation);
create_undistortion_lut(&cameras[ii].calibration, K4A_CALIBRATION_TYPE_COLOR, &pinhole, IR_lut, IR_interpolation);

k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(uint16_t), &undistorted_depth_image);
k4a_image_create(K4A_IMAGE_FORMAT_COLOR_BGRA32, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(uint32_t), &undistorted_color_image);
k4a_image_create(K4A_IMAGE_FORMAT_IR16, pinhole.width, pinhole.height,
				pinhole.width * (int)sizeof(uint16_t), &undistorted_IR_image);

color_image = k4a_capture_get_color_image(cameras[ii].recent_capture);
depth_image = k4a_capture_get_depth_image(cameras[ii].recent_capture);
IR_image = k4a_capture_get_ir_image(cameras[ii].recent_capture);

transformed_depth_image = depth_to_color(cameras[ii], depth_image, color_image);
transformed_IR_image = IR_to_color(cameras[ii], depth_image, IR_image, color_image);

remap(transformed_depth_image, depth_lut, undistorted_depth_image, depth_interpolation);
remap(color_image, color_lut, undistorted_color_image, color_interpolation);
remap(transformed_IR_image, IR_lut, undistorted_IR_image, IR_interpolation);`

I have omitted variable initialization, and I can provide depth_to_color or IR_to_color is requested, but they just (successfully) transform those respective images to color space, preserving 16 bit data in the color camera's coordinates/resolution.

The "non-undistorted" color image:
[image]

The "undistorted" color image:
[image]

rabbitdaxi, 2020-03-04:
may be this can help you https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/undistort

SeanStephensenAIRM, 2020-03-04:
> may be this can help you https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/undistort

unfortunately not. As I mentioned in the original post, that example code was the basis of the code that I have shared. That example code only undistorts a depth image, which is the only one that worked for me. The example doesn't seem to give any other suggestions for different parameters required to undistort color and IR images, from what I can tell, aside from pointing out that the calibration_type_t and interpolation_t would need to change, which I have done.

rabbitdaxi, 2020-03-09:
You are correct. The undistort example indeed is just an example to show case how to undistort a depth image. The principle of how to compute the undistortion lut can be shared by any type of images (it is a concept only depends on the camera intrinsics), it computes the pixel indices mapping relationship between distorted and undistorted images. So you can reuse these code for your purpose. 

However, your issue could related to the remapping function. The example is just show case how to consume the LuT for depth image, therefore, the remapping function here assume a single channel uint16_t image. (therefore, the remapping should be usable for IR image too, which share the same type as depth). However, for color, you need to extend the remapping to support the BGRA32. Great suggestion on the enhancement of the undistortion example.

SeanStephensenAIRM, 2020-03-10:
>therefore, the remapping should be usable for IR image too, which share the same type as depth

The remapping function failed for me on the IR image. It did not fail for color images, but obviously the result was wrong, and as you suggest, would probably need adjustment. 

Unfortunately I have abandoned this code and moved any undistortion/point cloud generation to a separate program using opencv/o3d, so I can not provide any further debugging/error codes related to this issue. My above code would provide more insight for anyone interested - it calls unaltered functions from the undistortion example.

rabbitdaxi, 2020-05-01:
Thank you for confirming you find solution for your issue. The example in this repo was intend to just show a simple way of undistort depth and what should be pay attention to during undistortion, as described in the readme.md of the example. But good suggestion of extending it which is a good item for community support.

closed 2020-05-01 by rabbitdaxi