microsoft/Azure-Kinect-Sensor-SDK #1106: Vaguely human shapes generate body tracking skeleton

billpottle, 2020-03-05:
**Describe the bug**
When recording in our martial arts school, the body tracking sdk generates a skeleton for this Body Opponent Bag (BOB)
[image]

**To Reproduce**
You would need the pictured object to reproduce, but I am happy to share files. 

**Expected behavior**
I expect the SDK would use the fact that there is no lower body, no arms, no movement, and no difference from ambient room temperature to conclude that the object is not a living person. 

**Desktop (please complete the following information):**
 - OS with Version: Version	10.0.18362 Build 18362
 - SDK Version:  body tracking 1.0.0

**Additional context**
We can of course ignore the skeleton in post processing, but I believe this represents an area of improvement for the SDK. 

Certainly, the SDK should take into account that people could be missing various parts of their bodies. However, the combination of all the factors, especially the lack of body heat, should be enough to generate a correct result.

SeanStephensenAIRM, 2020-03-05:
Azure Kinect doesn't have a temperature sensor, so it will have trouble detecting a presence or lack of distinct body temperature.

SeanStephensenAIRM, 2020-03-05:
@billpottle The Kinect sensor operates in the NIR region, specifically at 850nm. Thermal infrared is transmitted at 8000-15000 nm, far above the range of the Kinect.

qm13, 2020-03-11:
The DNN has no way of knowing if the detected person is real or not. It also cannot determine if the person (or training dummy in this case) has no arms of that the arms are obscured. We are investigating options to reduce false positives. We are currently collecting recordings of scenes with objects that could be confused as humans but no real humans in view. Could you capture a scene with the training dummy? If so reply with your email address and I will send you instructions on sharing the recording with Microsoft.

To work around the problem you could use the joint confidence numbers to potentially eliminate the skeleton.

billpottle, 2020-03-31:
@qm13 - Thank you. It's not a huge issue for us, but just wanted to do what we can to improve the tech. I already have signed the release from sharing with a previous issue, so I sent  a link to the recording directly to your email.

qm13, 2020-04-01:
@billpottle Thanks for the capture.