microsoft/Azure-Kinect-Sensor-SDK #1111: syntax error in simple_cpp_sample

devLupin, 2020-03-06:
I am testing simple_cpp_project.

Add sentences as needed.
#include <WinSock2.h>

Then an error occurred in k4a.hpp  that did not exist before.
Same as the attached picture
[image]

===========================[Error log]=============================
==============================================================
1>------ Build started: Project: simple_cpp_sample, Configuration: Debug x64 ------
1>main.cpp
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(56,96): warning C4003: not enough arguments for function-like macro invocation 'min'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(59,86): warning C4003: not enough arguments for function-like macro invocation 'max'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(64,54): warning C4003: not enough arguments for function-like macro invocation 'max'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(56,96): error C2589: '(': illegal token on right side of '::'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(56,96): error C2760: syntax error: unexpected token '(', expected ')'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\packages\Microsoft.Azure.Kinect.Sensor.1.3.0\build\native\include\k4a\k4a.hpp(56,96): error C2760: syntax error: unexpected token '(', expected ':'
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\MyJoint.h(59,25): warning C4091: 'typedef ': ignored on left of '<unnamed-enum-X>' when no variable is declared
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\main.cpp(115,20): warning C4018: '<': signed/unsigned mismatch
1>C:\Users\devLupin\Source\Repos\Azure-Kinect-Samples-master\body-tracking-samples\simple_cpp_sample\main.cpp(193,63): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
1>Done building project "simple_cpp_sample.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

What is the problem??

haimingZZ, 2020-03-16:
I met the same error;
1>F:\git_vcpkg\vcpkg\installed\x64-windows\include\k4a\k4a.hpp(56,96): error C2589: '(': illegal token on right side of '::'
1>F:\git_vcpkg\vcpkg\installed\x64-windows\include\k4a\k4a.hpp(56,96): error C2760: syntax error: unexpected token '(', expected ')'
1>F:\git_vcpkg\vcpkg\installed\x64-windows\include\k4a\k4a.hpp(56,96): error C2760: syntax error: unexpected token '(', expected ':'

haimingZZ, 2020-03-16:
using 
#define NOMINMAX
can avoid this error

Abdul-Mukit, 2020-09-08:
Getting the same error. Tried to add k4a.hpp for creating a dll. 
@haimingZZ where do I insert the #define NOMINMAX ?

Abdul-Mukit, 2020-09-08:
Update:
I was able to fix it. Thanks, @haimingZZ.
I added the `#define NOMINMAX` before the line` #include <windows.h>` in my framework.h for creating my dll. 
The documentation provided for NOMINMAX is as follows:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/examples/viewer/opengl/k4aimgui_all.h#L17
> Some of these headers end up including windows.h, which defines the min/max macros, which conflict
> with std::min and std::max, which we're using because they're portable.  This macro modifies the
> behavior of windows.h to not define those macros so we can avoid the conflict.

Hope this helps whoever gets this problem in the future :)

closed 2020-09-11 by devLupin