microsoft/Azure-Kinect-Sensor-SDK #1112: replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    451066 type:Color

Elok1, 2020-03-06:
Hi,
I don't know how but sometime I have this error when I start processing frame.
I have it about once every 10 executions.
Error: 
> [2020-03-06 10:52:33.995] [error] [t=25188] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    251055 type:Color
[2020-03-06 10:52:34.075] [error] [t=25188] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    317733 type:Color

I'm actually coding in C#.

Thanks.

wes-b, 2020-03-09:
It means the depth stream is not keeping up with the color stream. It might be because your GPU can't handle the depth compute needed. If you capture a full log at info we might be able to tell you more. There are other warnings generated when GPU compute is not able to keep up.

Elok1, 2020-03-11:
thanks for the answer,

here is the log info: 

> New logging session started
[2020-03-11 11:22:03.340] [info] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (337): find_libusb_device(). Container ID found: {3d8795dc-2bcf-4b3d-4694-60032a485e76} 
[2020-03-11 11:22:03.342] [info] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000709593812
[2020-03-11 11:22:03.405] [info] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (337): find_libusb_device(). Container ID found: {3d8795dc-2bcf-4b3d-4694-60032a485e76} 
[2020-03-11 11:22:03.406] [info] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000709593812
[2020-03-11 11:22:03.434] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (278): ******************** Device Info ********************
[2020-03-11 11:22:03.434] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (279): K4A SDK version:     1.3.0
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (285): Serial Number:       000709593812
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (289): RGB Sensor Version:  1.6.102
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): Depth Sensor Version:1.6.75
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (295): Mic Array Version:   1.6.14
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (298): Sensor Config:       6109.7
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (301): Build type:          Release
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (306): Signature type:      MSFT
[2020-03-11 11:22:03.460] [critical] [t=23892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (312): ****************************************************
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (864): k4a_device_start_cameras(). Starting camera's with the following config.
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (865): k4a_device_start_cameras().     color_format:0
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (866): k4a_device_start_cameras().     color_resolution:1
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (867): k4a_device_start_cameras().     depth_mode:2
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (868): k4a_device_start_cameras().     camera_fps:2
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (869): k4a_device_start_cameras().     synchronized_images_only:1
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (870): k4a_device_start_cameras().     depth_delay_off_color_usec:0
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (871): k4a_device_start_cameras().     wired_sync_mode:0
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (872): k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[2020-03-11 11:22:05.783] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (873): k4a_device_start_cameras().     disable_streaming_indicator:0
[2020-03-11 11:22:05.805] [info] [t=4356] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dynlib\dynlib_windows.c (131): dynlib_create(). Depth Engine loaded C:\Users\xxxxxx\Desktop\STAGE\KinectV3\RoomManagementCore\bin\x64\Debug\depthengine_2_0.DLL
[2020-03-11 11:22:05.805] [info] [t=4356] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\deloader\deloader.c (32): verify_plugin(). Loaded Depth Engine version: 2.1.1
[2020-03-11 11:22:06.168] [info] [t=10420] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (912): k4a_device_start_cameras(). k4a_device_start_cameras started
[2020-03-11 11:22:07.111] [warning] [t=4356] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dewrapper\dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 913ms (this may be transient).
[2020-03-11 11:22:07.111] [info] [t=4356] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (170): queue_pop(). Queue "dewrapper" dropped oldest 22 captures from queue.
[2020-03-11 11:22:07.219] [error] [t=15268] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    117744 type:Color
[2020-03-11 11:22:07.251] [error] [t=15268] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    151044 type:Color
[2020-03-11 11:22:07.283] [error] [t=5252] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    184377 type:Color

wes-b, 2020-03-13:
> [2020-03-11 11:22:07.111] [warning] [t=4356] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dewrapper\dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 913ms (this may be transient).

This indicates your GPU is not performing.  It needs to be less than 33ms if you don't want to be dropping frames.

closed 2020-03-13 by wes-b

AswinkarthikeyenAK, 2020-08-24:
> This indicates your GPU is not performing. It needs to be less than 33ms if you don't want to be dropping frames.

I am using jetson nano and I have the same error. 
Is there a way to parameterize this value such that I can avoid this error?

siraj-haid, 2020-09-14:
I have the same error when opening the Azure Kinect viewer. It is giving me the same error and then crashing when i choose the WFOV as a depth configuration.

Does anyone know anything about it?
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1041066 type:Color

zhuoyuan00, 2023-05-18:
Hi I am using a Intel NUC13 i7 and met the same problem. How can I deal with it?