microsoft/Azure-Kinect-Sensor-SDK #1113: Offset in Skeletons from Azure Kinect and Kinect v2

justamad, 2020-03-07:
Hello, 

for my use case, I want to register the Skeletons from Azure Kinect to the Skeletons coming from Kinect v2. In the calibration process, I'm using an 8x5 50mm checkerboard and identify three corner points using OpenCV, and for each Kinect, using the mapper functions to go from 2D (x,y) color pixel location to 3D (x,y,z) camera space (see image 1,2)

[image]

[image]
  
After moving the checkerboard 4-6 times and capture new calibration images for both cameras, I'm using SVD to get a rotation and translation between the two camera locations. The mean squared error of the rotated points is usually less than 1cm. However, when applying the resulting rotation+translation onto the skeletons, I receive an offset in the y-axis (see image), when x and z-axis are looking good (see screenshot) 

[image]
 
System Setup Azure:
OS: Windows 10
GPU: Nvidia RTX 2060
RAM: 16GB
Azure Kinect SDK: 1.3.0
Body Tracking: 1.0.0

Kinect v2:
OS: Windows 10
Library: PyKinect2 (Python) (with Kinect Studio)

davidkitman, 2020-03-09:
Could this be related to the tilt of the depth camera on the Kinect Azure vs its RGB camera?
"The depth camera is tilted 6 degrees downwards of the color camera"
Are you transforming for that?
https://docs.microsoft.com/en-us/azure/kinect-dk/coordinate-systems

justamad, 2020-03-09:
Hi davidkitman,

thanks for your advice! I manually corrected for it by applying a rotation matrix of 6 degrees around the x-axis and now the result is looking much better, see the image below. 

[image]

closed 2020-03-09 by justamad

changikjun, 2021-04-05:
How did you offset the data Azure to v2?
Can I get the Sequence..?

justamad, 2021-04-08:
For the initial rotation+translation matrix between the checkerboards please check out the following links for description and the Python code:

http://nghiaho.com/?page_id=671
https://github.com/nghiaho12/rigid_transform_3D

After applying the rot+trans matrix, I corrected for the depth camera tilt by multiplying an Euler-Rotation matrix around the x-axis by 6 degrees to all joints over all frames, as shown below:

```python
import numpy as np

def create_rotation_matrix_x(angle=6):
    angle = angle * np.pi / 180
    rot_x = np.matrix([[1, 0, 0],
                       [0, np.cos(angle), -np.sin(angle)],
                       [0, np.sin(angle), np.cos(angle)]])
    return rot_x
```

changikjun, 2021-04-09:
Thanks for your advice!
But actually, I'm use c++..
And I have more question!
If I get data of Azure, (instrinsic, extrinsic, etc,,,)
How Can Do I insert in V2 code...??