microsoft/Azure-Kinect-Sensor-SDK #1119: Obtaining the distance of joints with respective to the ground plane

narmadha14, 2020-03-10:
Hi,

Is there any existing parameters in the SDK that could readily calculate the distance between the 3D joint coordinates and the ground plane? I believe in the older Kinects, you can use the FloorClipPlane from the SkeletonFrame to get the plane equation (X,Y,Z,W) for the floor. Then I could easily get the distance between a plane and a point using the Point-Plane distance formula. Is there an equivalent way to obtain this in the current Kinect Azure SDK? 

Thanks in advance!

davidkitman, 2020-03-10:
There is a a C++ sample for getting the floor plane equation from point cloud and IMU sampling.
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/floor_detector_sample

narmadha14, 2020-03-10:
Thanks for the reply David. I have already explored this FloorDetection sample and I haven't been successful in getting a decent estimation of the floor plane.  In my experiment, I have three Kinects setup from multiple viewpoints (all top view) to observe the 3D scene inside a room. I am trying to calculate the joint distances from the skeletal frame to the ground plane for each device respectively. 

The given sample code does calculate the ground plane based on the partial point cloud data it observed in the room. But the projection of the assumed ground plane (SetFloorRendering) onto the depth map isn't the actual floor. I am not sure how it must be modified to be actually representative of the ground. 

Please let me know if you have any insight as to how I can fix it. Much appreciated. 

Thanks!

narmadha14, 2020-03-11:
[image]

In the screenshot, you can see the three devices running floor detection. Each device calculates the ground plane individually. And I am printing the floor and normal parameters (p.X, p.Y, p.Z, n.X, n.Y, n.Z) returned from: 
`auto p = maybeFloorPlane->ProjectPoint(cameraOrigin) + maybeFloorPlane->ProjectVector(cameraForward) * 1.5f;` 
`  auto n = maybeFloorPlane->Normal;`

billpottle, 2020-03-11:
for my application I 'calibrated' the ground plane by assuming it was between the subject's ankles. It's just an estimation and I agree it would be helpful to have a better way.

davidkitman, 2020-03-12:
I have a horizontal camera setup (not pointing down as much as yours) but the flooring has poor IR reflectance. Adding a mat helped me a lot. You can see it in the gif in my case the estimation of the floor plane is quite good. It does not align / match up with the corners of the room if you need that but works for me. 
[image]

Could your environment be the cause of any misalignment.. clutter or lower horizontal planes in view (the lowest horizontal plane I believe is considered the floor)

abalanmsft, 2020-03-17:
> But the projection of the assumed ground plane (SetFloorRendering) onto the depth map isn't the actual floor.

For any single camera view, can you be more specific about how the estimated ground plane differs from the actual floor? By rotating the camera view in the 3D Visualization window, you may be able to get a better sense of what is going on.
- Is the plane estimate inconsistent (too jittery), or reasonably consistent but significantly offset in either elevation or tilt or both?
- Is the plane elevation too high or too low? If so, by how much? Is it within the noise of the depth pixels? Are there phantom depth pixels that are lower than the actual floor?
- Is the estimated plane or the actual floor tilted in any way from vertical? If so, by how much? 
- Is any of the camera views better or worse than the others?

closed 2020-04-25 by qm13