microsoft/Azure-Kinect-Sensor-SDK #1125: Cuda 10.1 support

derivativerobmc, 2020-03-11:
Is there a timeline for updating the library to Cuda 10.1? We're switching over to Visual Studio 2019 and my understanding is that cuda 10.0 has problems with visual studio 2019.

yijiew, 2020-03-16:
All our development and sample code projects are using VS2019 and we didn't see any problem. Could you clarify what kind of issue you will hit by using Cuda 10.0 dll in VS2019?

derivativerobmc, 2020-03-17:
host_config.h in the Cuda 10.0.130 SDK has a line that checks for the MSC_VER and currently throws an error if it's greater than or equal to 1920 with the message that only VS 2013 - 2017 are supported

yijiew, 2020-03-18:
Thanks for clarification. But for the body tracking SDK, only the cuda binaries are needed. Customers shouldn't need to link to Cuda SDK at all.

derivativerobmc, 2020-03-18:
We're not using it for the body tracking, but we do use the Cuda SDK a lot in other parts of our project. We've disabled it for the moment to get things working, but we'll need it in the long run.

I noticed the onnxruntime 1.2 says it upgraded to cuda 10.1 about a week ago. Will this get carried over to the kinect relatively soon?

thanks for the help.

derivativerobmc, 2020-06-09:
Any word on updating the body tracking sdk to use a newer version of cuda? Other components in our project require 10.1 and we don't want to be shipping multiple versions.

maruta, 2020-10-06:
I'm experiencing a problem with k4abt_simple_3d_viewer.exe on a Windows machine with RTX 3080, where the tracking window doesn't appear and not even a log file is generated.

I'm beginning to suspect that a lower version of the bundled CUDA is the cause, as it works when I remove the RTX 3080 and install a much older GT710, and it works even with the CPU options specified.

For your information, the same binary works correctly on another Windows machine with Quadro RTX 5000 (Turing architecture).

So far, I'm not sure if that's a reason to upgrade the CUDA version or if it's a problem with my RTX3080 unit, I'm writing here in the hopes that someone else experiencing the same phenomenon will find it.

qm13, 2020-10-06:
We have had several reports of issues with the RTX 3080. The issue is likely with CUDA version which is an ONNX Runtime dependency. We are working with the ONNX team to make sure this is addressed in the next release of the Body Tracking SDK which is targeted for end of October / early November.

chrisnovello, 2020-10-28:
+1 having the same problem with k4abt_simple_3d_viewer.exe as @maruta, also on rtx 30 series, also had same binary working on a different machine & gpu. runs (very slowly) on this current machine if I execute with the CPU only mode

additionally, seeing a freeze upon trying to init the unit with an Unreal plugin that uses the sdk, which might be related (same plugin was also working on other machine)

eager for new release and to see if 30 series helps give a boost to skeleton tracking

datee, 2020-11-17:
Can confirm this issue also, on a computer with RTX 3090 the k4abt_simple_3d_viewer does not open the tracking window. 
It just stops. No crash but just nothing happening. CPU mode works fine.

When compiling and testing the examples **(simple_cpp_sample)** ..its the same.. stops at this line : 
` k4abt::tracker tracker = k4abt::tracker::create(sensor_calibration);`

Really need this, hope a fix is on the way! :)

L4Z3RC47, 2020-11-17:
> Can confirm this issue also, on a computer with RTX 3090 the k4abt_simple_3d_viewer does not open the tracking window.
> It just stops. No crash but just nothing happening. CPU mode works fine.
>
> When [...]

Confirmed here as well. RTTX 3090. 

@yijiew any updates on ETA for the next release?

UEBoy2019, 2020-11-29:
Who can tell me, has this problem been solved and what needs to be updated. I updated to cudn10.1, but it has no effect.
Do I need to move the CUDN DLL library to bianary replacement of body tracking file ？？
#1423

RoseFlunder, 2020-11-29:
I think Microsoft needs to release a new Version thats compiled with a Onxx runtime that uses cuda 11.
I think we as users can just wait.
Is that right?

qm13, 2020-11-30:
Currently investigating issue.

L4Z3RC47, 2020-12-13:
> 
> 
> We have had several reports of issues with the RTX 3080. The issue is likely with CUDA version which is an ONNX Runtime dependency. We are working with the ONNX team to make sure this is addressed in the next release of the Body Tracking SDK which is targeted for end of October / early November.

@qm13 is there a new target date for the next release?

damaza, 2020-12-25:
> I'm experiencing a problem with k4abt_simple_3d_viewer.exe on a Windows machine with RTX 3080, where the tracking window doesn't appear and not even a log file is generated.
>
> I'm beginning to [...]

The same problem with RTX3070

L4Z3RC47, 2020-12-25:
All 3000 series cards are affected. Hoping we can get an ETA soon. I have commercial applications that could really use this!!!

UEBoy2019, 2020-12-28:
> Currently investigating issue.

  i am also very worried. Please ,please tell me the completion data !
3Q

jenyaone, 2020-12-30:
> Currently investigating issue.

are there any updates on this? please let us know.

Lt23live, 2021-01-05:
We are also having this issue, RTX 3080 series. Microsoft needs to address this.

L4Z3RC47, 2021-01-05:
ping @qm13

Lt23live, 2021-01-07:
qm13 has there been nay updates?

AquaLamp, 2021-01-10:
Thank you. I am eagerly awaiting the results of your investigating issue.

damaza, 2021-01-11:
@qm13

L4Z3RC47, 2021-01-12:
@qm13 any news? This ticket seems to revolve around onnxruntime support of CUDA 10.1 however onnxruntime has supported both cuda 10.2 and cuda 11 since release 1.5.1 
I assume that this is more than just a simple runtime issue then but given this is a major blocker on current gen hardware it would be great to get some deeper feedback on what the issue investigation has revealed and what/if there is a timeline for an update. 
Thank you!

UnaNancyOwen, 2021-01-19:
@qm13 Currently, the market is moving to GeForce 30XX generation products, and GeForce 20XX/10XX generation products have come to end of sales. NVIDIA has announced the 30XX generation products for laptops. All desktop/laptop on the market will soon be replaced by 30XX generation GPU.
This is a very serious problem for businesses. In fact, we are having trouble procuring new PC that deliver to our customers for work. Please, hurry up with this task (update ONNX Runtime to CUDA 11.x, or support [DirectML provider](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1343)). Thanks!

L4Z3RC47, 2021-01-20:
> @qm13 Currently, the market is moving to GeForce 30XX generation products, and GeForce 20XX/10XX generation products have come to end of sales. NVIDIA has announced the 30XX generation products for [...]

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481#issuecomment-763288112

Per @qm13 looks like we should see an update around mid next month (usual caveats apply)

great news!

jaymimc, 2021-03-17:
> We have had several reports of issues with the RTX 3080. The issue is likely with CUDA version which is an ONNX Runtime dependency. We are working with the ONNX team to make sure this is addressed in the next release of the Body Tracking SDK which is targeted for end of October / early November.

Sorry, know i have also listed this on another ticket but it suggested it was closed. As this is still an open ticket, is there any update on when SDK, CUDA and Body Tracking working with NVIDIA 3xxx cards?

qm13, 2021-03-19:
Upgraded to CUDA 11.1

closed 2021-03-19 by qm13

AswinkarthikeyenAK, 2021-03-30:
Hi Guys,
Can anyone confirm if the body tracking is working fine without any issues with the RTX30xx graphic card,please?
I am planning to buy a desktop with the RTX3070 and I wanted check before I could purchase the desktop. 
Thanks for your feedback !!

lewkoo, 2021-03-30:
I can confirm that it works fine on RTX 3070 whereas before it didn't. 
See screenshot below as proof. 
[image]

AswinkarthikeyenAK, 2021-03-30:
Hi @lewkoo,
Thanks for the information

AhmedMedhat, 2021-04-18:
After upgrading the body tracking library to v1.1, the `k4abt_simple_3d_viewer` example runs with the CUDA option, however printing the following message:
``` 
2021-04-17 21:15:30.444827715 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-04-17 21:15:30.446831171 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
```
What does this message mean? @lewkoo did you get similar message when running on RTX 3070?
Thanks,

lewkoo, 2021-04-20:
> After upgrading the body tracking library to v1.1, the `k4abt_simple_3d_viewer` example runs on RTX 3090 (CUDA Version: 11.2), however it prints the following message before creating the tracking window:
> 
> ```
> 2021-04-17 21:15:30.444827715 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
> 2021-04-17 21:15:30.446831171 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
> ```
> 
> @qm13 What does this message mean? Does it fallback to CPU processing? @lewkoo did you get similar message when running on RTX 3070?
> Thanks for the efforts,

I got identical message when running `k4abt_simple_3d_viewer` on 18.04 Ubuntu with CUDA 11.2. Possibly due to [this](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1481#issuecomment-803077212).

AFAIK, it doesn't fallback to CPU processing, at least not entirely. `nvidia-smi` still shows a process and GPU is experiencing some additional load. From the error message, I can wild guess that it is off loading body shape detection to CPU, while everything else is still running on GPU but this is just a guess. However, our applications that use body tracking seem to perform just as they used to despite these error messages.

qm13, 2021-04-26:
@lewkoo That warning is expected. Please see the troubleshooting notes for AKDK in the documentation.

ZodiacWind, 2021-12-09:
> I have a question. I use Win10P RTX3080 to turn on body tracking v1.1.0 but it can’t be displayed. I use CUDA 11.1 according to the article. Is anyone like me? I am a student first contact with Azure Kinect

I have encountered the same issue. Could anybody provide some instructions? Really appreciate it.

VegetableWithChicken, 2021-12-09:
您好，我一般不常用邮箱，有重要邮件，请QQ我下，谢谢!

stroombeestje, 2022-02-19:
> I have encountered the same issue. Could anybody provide some instructions? Really appreciate it.

Same issue here

tinywolf3, 2023-06-11:
same here. please fix it.