microsoft/Azure-Kinect-Sensor-SDK #1127: azure kinect drops frames

yellowfisher, 2020-03-12:
I've record color and depth stream time stamp, the nearest two frames have 100ms interval，that means 3 frame drops, is this normal?
[image]

wes-b, 2020-03-12:
There might be a couple when starting the stream, but after the first second I would expect no drops from a PC that meets the performance specs.

yellowfisher, 2020-03-12:
> There might be a couple when starting the stream, but after the first second I would expect no drops from a PC that meets the performance specs.

thanks for your quick reply, as picture shows, it drops from the frame 1482. My computer has i7-9700k cpu, 16gb memory, 1T ssd and nvdia rtx 2080 graphics card.

wes-b, 2020-03-12:
Please use our bug template to create a more complete request (you can enhance this issue). No way to know with so little details.

yellowfisher, 2020-03-12:
> Please use our bug template to create a more complete request (you can enhance this issue). No way to know with so little details.

I'll do a deep test and create complete request later. There's another problem here, i get skeleton data with `k4abt_frame_get_body_skeleton`, map 3d position data to 2d rgb data with `k4a_calibration_3d_to_2d`, and at last draw skeleton with opencv `cv::line` , but the skeleton just out of body, did i do it wrong?
[image]

wes-b, 2020-03-12:
I would say if you are going to do a new test and you suspect an issue. Try with one of our tools first.  streaming_samples.exe and Throughput_perf are useful for verifying everything works as expected.

wes-b, 2020-03-30:
Let us know if you have new information about this issue.

closed 2020-03-30 by wes-b

yellowfisher, 2020-09-25:
> Let us know if you have new information about this issue.

I truly shure azure kinect drop frames, i test with your recorder tool. I suggest you do a test yourselves.