microsoft/Azure-Kinect-Sensor-SDK #1128: Applying the body tracking data (position and quaternion) to my own human model

haeginh, 2020-03-12:
Hi, I am having a difficulty in understanding the position and orientation concepts given for each joint of the skeleton. What I am trying is to put head and hand models in right orientation and right position, according to the acquired data, which have been done before by using other motion capture devices that support BVH files.

What first came up to me was that I could simply rotate the head model according to the quaternion and translate it to the corresponding position. However, the problem was that I don't know the initial position or orientation of the head model that is supposed to be (I don't know the rest pose).

Could you please kindly show the brief procedure to do this?

Plus, according to my understanding, quaternion is rotation information rather than orientation. However, Kinect SDK provides the orientation of the joints as quaternion. Is there any base vector you are assuming? Like (1, 0, 0), so that I can get the orientation by calculating (rotation matrix)*(1, 0, 0)?

knat, 2020-03-21:
[CharacterAnimator](https://github.com/bibigone/k4a.net/blob/master/K4AdotNet.Samples.Unity/Assets/Scripts/CharacterAnimator.cs) demonstrates how to use the quaternions to drive a avatar.

haeginh, 2020-04-02:
Thank you very much. I will go through the codes in the link. Thanks :-)

closed 2020-04-14 by qm13