microsoft/Azure-Kinect-Sensor-SDK #1129: Possibly distorted space in Azure Kinect

B747-400, 2020-03-13:
Hi,

We use Kinects for Azure and want to calibrate one of our applications with a 10x7 chessboard. After a few unsuccessful attempts we decided to check point cloud generation in Kinects for Azure SDK “transformation_example” application (this sample application aligns depth image to color, generates the point cloud and exports it to PLY). Results are not satisfactory: chessboard squares become wider when moving from the center of the board. Differences in squares dimensions reach up to 10 mm. Is it possible that the “transformation_example” application generates distorted point clouds or we see some space distortion for some other reason?

Generated PLY files and measurements screenshot are placed at https://drive.google.com/open?id=1Z0aIqMHNt2mvMHFATYwIkiCsJLD-pBvU

Thanks in advance,
Virgis

rabbitdaxi, 2020-04-24:
@B747-400 the error could coming from a few things
- the intrinsics and extrinsics comparing to the hardware current status
- the depth error, since during 2d to 3d, depth is used, then the depth error could contribute to x, y, z direction, please refer to the [spec](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes)

rabbitdaxi, 2020-05-18:
close after 7 days policy

closed 2020-05-18 by rabbitdaxi