microsoft/Azure-Kinect-Sensor-SDK #1130: Documentation for bodytracking has incorrect examples for realtime vs synchronous processing

Chris45215, 2020-03-13:
As per the guidance from @yijiew in issue 816: "As for the additional queued frame, if you care the latency so much, you should always use synchronized call (set timeout to K4A_WAIT_INFINITE for both enqueue and pop result function). Then you will guarantee to get the result of the frame you pass in." 

This contradicts the documentation at https://docs.microsoft.com/en-us/azure/kinect-dk/get-body-tracking-results. That documentation says under Real Time Processing that "real-time results and can accommodate dropped frames" should use 0 timeout. Whereas for Synchronous Processing in "applications that do not need real-time results or cannot accommodate dropped frames", it instructs to use K4A_WAIT_INFINITE.

So, it appears that the documentation has it backwards, or something needs to be changed.

yijiew, 2020-03-13:
The documentation actually is correct under the context of Body Tracking SDK. Because if you set timeout to be 0, it means when you feed in a frame to the tracker, it might fail. Then the frame will be dropped. So you should only set timeout to be 0 when you can "accommodate dropped frames". The word "real time results" might be a bit confusing here. Actually, the real-time result here means the tracker will process faster if you keep queuing the frame without waiting for previous frame to be finished. Because then the pipeline can process two frames at different stages at the same time. But it doesn't mentioned any latency here. If you care about the latency and still want the fast processing to happen, you could do something like I mentioned in the other post: "implement a while loop, to keep calling k4abt_tracker_pop_result function with timeout to be 0 until it fails. The last result you could get is the latest. You can ignore the older stuff on your own code."

Chris45215, 2020-06-15:
@yijiew, the description for the real-time body tracking says "Use this pattern for single-threaded applications that need real-time results and can accommodate dropped frames." 

So I find it odd that this example is one that doesn't drop that frame at the start. I cannot think of a use-case where you would want real time processing (of any definition of "real time") and can tolerate dropped frames, but prefer to have significantly higher latency through the life of the program by not dropping such a frame. As far as I can tell, the real difference between the code examples is merely which component waits for the other. The real-time example (supposedly) provides whatever result it has available if it has one (I have some tough questions for how it generates 2 frames in a single iteration if the program is supposed to iterate faster that the camera and DNN make frames, but that's for another time), and doesn't delay overall execution; while the synchronous one causes the overall program to wait until a frame is ready. I cannot think of a use case where you want the program to run asynchronously, but care so much about that first frame that you are unwilling to drop it. I suggest that either a 3rd example be added for the low-latency use case, or the real-time example be amended to drop extraneous frames. 

As it currently is, I don't see how it's actually intended to work - you always get the latest frame (if any) from the sensor, but do NOT get the latest frame from the tracker. Why would you want that?  That's like going to the grocery store, purchasing the ripest bananas you can find, then leaving them in the kitchen for 3 days so they can go bad.

Chris45215, 2020-07-11:
Why is this being closed when the code examples provide an unnecessarily latent result, and don't ever mention that fact or the workaround? Whatever you use the tracker for, if it's used for live/real-time/interactive (whatever term you want to use) programs, then you want low latency. So the how-to guide and code examples should provide optimal code for the best result possible, which means they should include the fix (using the While loop), or at least mention it. 

Alternatively, the asynchronous guide and examples should show the queue and dequeue happening in separate threads, so those processes can be fully async - if that actually improves performance or latency. It's a disservice to use an example that only goes halfway.

qm13, 2020-07-14:
@Chris45215 my bad. Missed you reply to @yijiew.

qm13, 2022-05-25:
Fixed in latest documentation coinciding with v1.1.2 release

closed 2022-05-25 by qm13