microsoft/Azure-Kinect-Sensor-SDK #1131: Would like depth_point to xyz_point conversion

ameyades, 2020-03-13:
**Is your feature request related to a problem? Please describe.**

Would like to convert a singular depth image point to a point cloud point.  k4a_transformation_t_get_context is not defined anywhere in the SDK, so I cannot find the xy_tables necessary to do the conversion by myself

**Describe the solution you'd like**

A  function that converts depth to xyz for a single point.

**Describe alternatives you've considered**

Being able to access the xy_tables or the transformation context by being able to call k4a_transformation_t_get_context in application code. 

**Additional context**

Doing this to map a transformed depth image point to a point cloud point that is being rendered in VTK.

RoseFlunder, 2020-03-13:
Maybe this example will help you:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/fastpointcloud/main.cpp

It shows how to create a xy-table which you can then use.

ameyades, 2020-03-14:
Thank you! However, still not clear about one thing: If my input is a depth image produced by k4a_transformation_depth_image_to_color_camera, what is the proper calibration to use to get those x and y points in the point cloud?

rabbitdaxi, 2020-04-24:
@ameyades if you just want to unproject single point from 2d to 3d using depth, have you tried [k4a_calibration_2d_to_3d()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga664602bdb48dab38117a6c1d14b880de.html#ga664602bdb48dab38117a6c1d14b880de)

rabbitdaxi, 2020-05-01:
@ameyades hope my answer above resolved your question. The SDK should already support you to do any transformation between 2d and 3d, between cameras, and also for both single point or the entire image. Recommend you can read the following docs:
- https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions
- https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation

Now, if you for some reason and really want to have access to xytables, the sdk open sourced code can allow you to do that. Take a look this function [implementation ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2494b92e2ee2b5b315cb38980b8aae5226d5e8d3/src/transformation/transformation.c#L465)

closed 2020-05-01 by rabbitdaxi

ameyades, 2020-05-01:
Hi,

Thank you for the help!

Very sorry for the late reply - currently working on another project so haven't had a chance to verify. I will test the function implementation when I get the chance.