microsoft/Azure-Kinect-Sensor-SDK #1134: Consistent shut down for all 3 Azure Kinect DK

datajoker1, 2020-03-18:
**Describe the bug**
We bought 3 Azure Kinect DKs.
When we start k4aviewer, all 3 will shut down after sending those error codes that i have listed below. 
However, if I turn the depthmode off and start the k4aviewer, it runs normally.

**Logs**
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    151044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    184377 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    217711 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    251044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    284388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    317722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    351055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    384400 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    417711 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    451055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    484377 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    517722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    551044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    584388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    617711 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    651055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    684388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    717722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    751044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    784388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    817722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    851055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    884377 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    917711 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    951055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    984388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1017722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1051066 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1084388 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1117722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1151044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1184377 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1217711 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1251044 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1284377 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1317722 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1351066 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1384400 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1417733 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1451077 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1484411 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1517711 type:Color
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1551055 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1584377 type:Color
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version:  Ubuntu18.04 LTS
 - SDK Version:  1.3.0
 - Firmware version: 

Listed below are the information for the firmware version and the serial numbers which matches exactly with 3 of our kinect dks
Device Serial Number: [000068792512, 000227392912, 000446792312]
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

wes-b, 2020-03-18:
> [ warning ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth.
[ error ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[ error ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063

You tried to push too much data through your USB Host controller. Eventually one of the requests timed out and failed. Try running 3 at 5 FPS and then stepping up to 15 and finally 30 FPS

datajoker1, 2020-03-20:
Thank you for your reply. I am only using one USB connector. I don't understand how and why this causes data loss. Using 3 or 5 frames is meaningless to me, so what is the reason for this happening.

wes-b, 2020-03-20:
Sorry I thought you were running 3 camera's at the same time. 

The point of my steps are to start small (slow), verify your system works as expected at the lowest setting, and then start increasing performance needs. Are you using the cables and power supply that came with the camera?

datajoker1, 2020-03-23:
yes we are using the cables that came with the camera. However, we sometimes use the extension cables(2m ~ 3m), but without it also causes the same error.

wes-b, 2020-03-23:
How did verification at slower speeds go? Do the camera's work any better on another PC? 

It is very unlikely you received 3 bad camera's so we need to look deeper at what issue might be in your setup. 

Try to stay away from the extension cables until you find a setup that works as you expect it to.

wes-b, 2020-03-30:
@datajoker1 closing this due to a lack of progress. Happy to reopen if you have new info.

closed 2020-03-30 by wes-b