microsoft/Azure-Kinect-Sensor-SDK #1135: How to install Azure-Kinect-Sensor-SDK and Azure_Kinect_ROS_Driver for Ubuntu 18 - ros Melodic

tuan97anton, 2020-03-18:
**I want to question : Do I install for true ? Did I skip any step ? 
I install follow below step :** 
### Install Azure-Kinect-Sensor-SDK
1 :    ```
        sudo apt-get update && sudo apt-get install -y \
        pkg-config \
        ninja-build \
        doxygen \
        clang \
        gcc-multilib \
        g++-multilib \
        python3 \
        git-lfs \
        nasm \
        cmake \
        libgl1-mesa-dev \
        libsoundio-dev \
        libvulkan-dev \
        libx11-dev \
        libxcursor-dev \
        libxinerama-dev \
        libxrandr-dev \
        libusb-1.0-0-dev \
        libssl-dev \
        libudev-dev \
        mesa-common-dev \
        uuid-dev \
        libopencv-dev
```

2 :    ```
        sudo apt install libk4a1.3
        sudo apt install libk4a1.3-dev
        sudo apt install k4a-tools
```

3: ```
     git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK
     mkdir build && cd build
     cmake .. -GNinja
     ninja
     sudo ninja install
```
### install Azure_Kinect_ROS_Driver
1 :```
     git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver
     cd catkin_ws/ 
     catkin_make
``` 

After I install follow above : I have issue when I `catkin_make` to install Azure_Kinect_ROS_Driver : 
```
Base path: /home/sti/catkin_ws
Source space: /home/sti/catkin_ws/src
Build space: /home/sti/catkin_ws/build
Devel space: /home/sti/catkin_ws/devel
Install space: /home/sti/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/sti/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/sti/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/sti/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/sti/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/sti/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: azure_kinect_ros_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - azure_kinect_ros_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'azure_kinect_ros_driver'
-- ==> add_subdirectory(Azure_Kinect_ROS_Driver)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
Finding K4A SDK binaries
-- Body Tracking SDK found: compiling support for Body Tracking
K4A Libs: k4a::k4a;k4a::k4arecord;k4abt::k4abt
K4A DLLs: /usr/local/lib/libk4a.so.1.4.0;/usr/local/lib/libk4arecord.so.1.4.0;/usr/lib/libk4abt.so.1.0.0
K4A Install Needed: FALSE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sti/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/sti/catkin_ws/build"
####
Scanning dependencies of target azure_kinect_ros_driver_nodelet
Scanning dependencies of target azure_kinect_ros_driver_node
[ 30%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_bridge_node.cpp.o
[ 30%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_bridge_nodelet.cpp.o
[ 30%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device.cpp.o
[ 40%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device_params.cpp.o
[ 50%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o
[ 60%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_calibration_transform_data.cpp.o
[ 70%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_ros_device_params.cpp.o
[ 80%] Building CXX object Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_nodelet.dir/src/k4a_calibration_transform_data.cpp.o
[ 90%] Linking CXX shared library /home/sti/catkin_ws/devel/lib/libazure_kinect_ros_driver_nodelet.so
[100%] Linking CXX executable /home/sti/catkin_ws/devel/lib/azure_kinect_ros_driver/node
[100%] Built target azure_kinect_ros_driver_nodelet
/usr/bin/ld: warning: libk4a.so.1.3, needed by /usr/lib/libk4abt.so.1.0.0, not found (try using -rpath or -rpath-link)
CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o: In function `cv::Mat::Mat(int, int, int, void*, unsigned long)':
k4a_ros_device.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x296): undefined reference to `cv::Mat::updateContinuityFlag()'
collect2: error: ld returned 1 exit status
Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/build.make:215: recipe for target '/home/sti/catkin_ws/devel/lib/azure_kinect_ros_driver/node' failed
make[2]: *** [/home/sti/catkin_ws/devel/lib/azure_kinect_ros_driver/node] Error 1
CMakeFiles/Makefile2:519: recipe for target 'Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all' failed
make[1]: *** [Azure_Kinect_ROS_Driver/CMakeFiles/azure_kinect_ros_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

roelofvandijk, 2020-03-19:
Looks like an OpenCV error. Can you check what OpenCV version you have? You might need a newer verison.

cv::Mat::updateContinuityFlag() was added somewhere after 3.3.1.

See also [this link](https://answers.opencv.org/question/198488/undefined-reference-to-cvmatupdatecontinuityflag/).

roelofvandijk, 2020-03-19:
This might actually be a bug - the dockerfile and the [setup-ubuntu.sh](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/scripts/docker/setup-ubuntu.sh) script indeed use `sudo apt-get install libopencv-dev` which installs an OpenCV version which might be too old (seems to be 3.2.0 per [Ubuntu](https://packages.ubuntu.com/eoan/libopencv-dev)).

tuan97anton, 2020-03-20:
I must use version OpenCV for install Azure kinect ?  Can you tutorial for me ?

tkircher, 2020-03-22:
The default Ubuntu 18.04 installation has unresolvable linking issues with ROS and OpenCV, so the Kinect ROS driver can't be built on a clean system. You have to not only build OpenCV-3.4.9 from source, but ROS melodic as well.

tkircher, 2020-03-22:
You'll want to install some prerequisites:

```
$ sudo apt-get install -y doxygen \
gfortran \
hdf5-tools \
jasper \
libatlas-base-dev \ 
libavcodec-dev \
libavformat-dev \
libblas-dev \
libbz2-dev \
libceres-dev \
libfltk1.3-dev \
libfltk-cairo1.3 \
libfltk-gl1.3 \
libgazebo9-dev \
libgpgme-dev \
libgpgme++2v5 \
libgtk-3-dev \
libgtkglext1-dev \
libhdf5-dev \
libhdf5-serial-dev \
libjpeg_dev \
liblapack-dev \
liblz4-dev \
liboctomap-dev \
libpng-dev \
libpoco-dev \
libqwt-qt5-dev \
libsdl-ttf2.0-dev \
libsoundio-dev \
libswscale-dev \
libtar-dev \
libtiff-dev \
libtinyxml-dev \
libudev-dev \
liburdfdom-dev \
libv4l-dev \
libvtk6-dev \
libvtk6.3-qt \
libx264-dev \
libxvidcore-dev \
libyaml-cpp-dev \
libzip-dev \
pyqt5-dev \
python-gpg \
python-sip-dev \
x11-xserver-utils \
zlib1g-dev \
```

Then for OpenCV:

```
$ apt-get install
$ wget https://github.com/opencv/opencv/archive/3.4.9.tar.gz -O opencv-3.4.9.tar.gz
$ wget https://github.com/opencv/opencv_contrib/archive/3.4.9.tar.gz -O opencv_contrib-3.4.9.tar.gz
$ tar xvzf opencv-3.4.9.tar.gz
$ tar xvzf opencv_contrib-3.4.9.tar.gz
$ cd opencv-3.4.9
$ mkdir build
$ cd build
$ cmake ..
$ make install
```

To build and install ROS, follow the instructions here:

http://wiki.ros.org/Installation/Source

roelofvandijk, 2020-03-22:
@tuan97anton:
You don't need to build the SDK (and install all those prerequisites, your first step) if you want to use it on Ubuntu 18.04, `sudo apt install libk4a1.3 libk4a1.3-dev k4a-tools` should be enough.

I can confirm that the following works on a fresh Ubuntu 18.04. virtual machine (in VirtualBox) and builds the Azure Kinect ROS driver without error:

```
# install necessary tools
sudo apt update && sudo apt -y upgrade
sudo apt install curl

# install ROS following http://wiki.ros.org/melodic/Installation/Ubuntu
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt update
sudo apt install ros-melodic-desktop-full
sudo apt install python-rosinstall python-rosinstall-generator python-wstool build-essential python-rosdep
echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
source ~/.bashrc
sudo rosdep init
rosdep update

# install Azure Kinect SDK following https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md
curl https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
sudo apt update
sudo apt install -y libk4a1.3 libk4a1.3-dev k4a-tools

# clone Azure Kinect ROS drivers, make workspace
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone https://github.com/microsoft/Azure_Kinect_ROS_Driver
cd ..
catkin_make

# [optional] make Azure Kinect available without being root
# sudo curl https://raw.githubusercontent.com/microsoft/Azure-Kinect-Sensor-SDK/develop/scripts/99-k4a.rules >> /etc/udev/rules.d
```

tkircher, 2020-03-22:
@roelofvandijk I'm sure that works on x86, but not ARM.

roelofvandijk, 2020-03-22:
@tkircher Thanks for pointing that out, let's hope OP tries to install it on x86.
 I see that ARM support will be released in 1.4 of the SDK - exciting.

tkircher, 2020-03-22:
@roelofvandijk To be honest, I forgot that people use ROS on non-ARM platforms when I responded. ;)

tuan97anton, 2020-03-23:
@roelofvandijk Thank you very much ! Your tutorial very detail . I tried and it work .

closed 2020-03-23 by tuan97anton