microsoft/Azure-Kinect-Sensor-SDK #1138: 4 separate depth images were obtained, depth image in OpenCV is purple

k123jack, 2020-03-19:
It may have already been posted, but I got 4 separated depth images in a screen and the color is purple im cv::imshow(). The color image got by k4a_capture_get_color_image() from the same capture is correct. 

Code : 

#pragma comment(lib, "k4a.lib")
#include <k4a/k4a.h>
#include <opencv2/opencv.hpp>
// #include <opencv2/highgui.hpp>
// #include <opencv2/highgui/highgui.hpp>
using namespace cv;
#include <stdio.h>
#include <stdlib.h>

int main()
{
    uint32_t count = k4a_device_get_installed_count();
    if (count == 0)
    {
        printf("No k4a devices attached!\n");
        return 1;
    }

    // Open the first plugged in Kinect device
    k4a_device_t device = NULL;
    if (K4A_FAILED(k4a_device_open(K4A_DEVICE_DEFAULT, &device)))
    {
        printf("Failed to open k4a device!\n");
        return 1;
    }
    printf("device opened!\n");
    // Get the size of the serial number
    size_t serial_size = 0;
    k4a_device_get_serialnum(device, NULL, &serial_size);

    // Allocate memory for the serial, then acquire it
    char *serial = (char*)(malloc(serial_size));
    k4a_device_get_serialnum(device, serial, &serial_size);
    printf("Opened devicaaaaaaaae: %s\n", serial);
    free(serial);

    // Configure a stream of 4096x3072 BRGA color data at 15 frames per second
    k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    config.camera_fps       = K4A_FRAMES_PER_SECOND_15;
    config.color_format     = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    config.color_resolution = K4A_COLOR_RESOLUTION_720P;
    config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
    config.synchronized_images_only = true;
    // Start the camera with the given configuration
    printf("start capturing\n" );
    k4a_capture_t capture;
    if (K4A_FAILED(k4a_device_start_cameras(device, &config)))
    {
        printf("Failed to start cameras!\n");
        k4a_device_close(device);
        return 1;
    }

    switch (k4a_device_get_capture(device, &capture, 10000))
    {
    case K4A_WAIT_RESULT_SUCCEEDED:
        break;
    case K4A_WAIT_RESULT_TIMEOUT:
        printf("Timed out waiting for a capture\n");
        // continue;
        break;
    case K4A_WAIT_RESULT_FAILED:
        printf("Failed to read a capture\n");
        // goto Exit;
        break;
    }

    printf("aaa\n" );

//  color image
    k4a_image_t color_image = k4a_capture_get_color_image(capture);
    if (color_image != NULL)
    {
        printf("aaa\n" );
        printf(" | Color32 res:%4dx%4d stride:%5d\n",
            k4a_image_get_height_pixels(color_image),
            k4a_image_get_width_pixels(color_image),
            k4a_image_get_stride_bytes(color_image));
        // 　↑　表示の部分
    // Release the image
    }
    k4a_image_t depth_image = k4a_capture_get_depth_image(capture);
    if (depth_image != NULL)
    {
        printf(" | Depth16 res:%4dx%4d stride:%5d\n",
                k4a_image_get_height_pixels(depth_image),
                k4a_image_get_width_pixels(depth_image),
                k4a_image_get_stride_bytes(depth_image));
    }

    printf("color depth image loaded\n" );
    k4a_image_format_t color_image_format=k4a_image_get_format(color_image);
    uint8_t* color_image_buffer = k4a_image_get_buffer(color_image);

    int color_image_rows = k4a_image_get_height_pixels(color_image);
    int color_image_cols = k4a_image_get_width_pixels(color_image);
    std::cout<< color_image_rows << color_image_cols << std::endl;
    cv::Mat color_image_colorMat(color_image_rows,color_image_cols,CV_8UC4,(void*)color_image_buffer,cv::Mat::AUTO_STEP);
    cv::imshow("Image", color_image_colorMat);
    cv::waitKey(0);
    cv::imwrite("color_image.jpg",color_image_colorMat);
    //show color image
    //↑get color image param
    // bool get_depth_image=false;

    k4a_image_format_t depth_image_format=k4a_image_get_format(depth_image);
    uint8_t* depth_image_buffer = k4a_image_get_buffer(depth_image);

    int depth_image_rows = k4a_image_get_height_pixels(depth_image);
    int depth_image_cols = k4a_image_get_width_pixels(depth_image);
    cv::Mat depth_image_colorMat(depth_image_rows,depth_image_cols,CV_8UC4,(void*)depth_image_buffer,cv::Mat::AUTO_STEP);
    cv::imshow("Image", depth_image_colorMat);
    cv::waitKey(0);
    cv::imwrite("depth_image.jpg",depth_image_colorMat);

    k4a_calibration_t calibration;
    //
    if (K4A_RESULT_SUCCEEDED==k4a_device_get_calibration (device,config.depth_mode,config.color_resolution,&calibration))
    {
        printf("calibrate succeed\n");

    }

    // if(k4a_calibration_2d_to_3d(calibration,))

    k4a_transformation_t transformation=k4a_transformation_create(&calibration);

    k4a_image_t output_image =NULL;
    if (K4A_RESULT_SUCCEEDED==k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16,color_image_cols,color_image_rows,color_image_cols*(int)sizeof(uint16_t),&output_image))
    {
        printf("create output_img succeed\n");
    }

    if (K4A_RESULT_SUCCEEDED==k4a_transformation_depth_image_to_color_camera(transformation,depth_image,output_image))
    {
        printf("compute transformed depth image succeed\n");
    }
    // k4a_image_format_t output_image_format=k4a_image_get_format(output_image);
    uint8_t* output_image_buffer = k4a_image_get_buffer(output_image);
    //
    // int output_image_rows = k4a_image_get_height_pixels(output_image);
    printf("rows get\n" );
    // int output_image_cols = k4a_image_get_width_pixels(output_image);
    cv::Mat output_image_colorMat(color_image_rows,color_image_cols,CV_8UC4,(void*)output_image_buffer,cv::Mat::AUTO_STEP);
    cv::imshow("Image", output_image_colorMat);
    cv::waitKey(0);
    cv::imwrite("output_image.jpg",output_image_colorMat);

    // Shut down the camera when finished with application logic
    k4a_device_stop_cameras(device);
    k4a_device_close(device);

    return 0;
}

4 separated depth Image :

[image]

Thx.

wes-b, 2020-03-19:
The depth image is an array of UINT16's that represent the distance in mm of the object the IR laser was reflected off of. So you can't display it as an image without converting the depth to a pixel.

k123jack, 2020-03-19:
how?

k123jack, 2020-03-19:
> The depth image is an array of UINT16's that represent the distance in mm of the object the IR laser was reflected off of. So you can't display it as an image without converting the depth to a pixel.

I found ColorizeDepthImage here :  examples/viewer/opengl/main.cpp

but I didn't find it in the API document.

eshirima, 2020-03-19:
@k123jack The depth image is a single channel thus change `cv::Mat output_image_colorMat(color_image_rows,color_image_cols,CV_8UC4,(void*)output_image_buffer,cv::Mat::AUTO_STEP);` to `cv::Mat output_image_colorMat(color_image_rows,color_image_cols,CV_8UC1,(void*)output_image_buffer,cv::Mat::AUTO_STEP);`

k123jack, 2020-03-19:
> @k123jack The depth image is a single channel thus change `cv::Mat output_image_colorMat(color_image_rows,color_image_cols,CV_8UC4,(void*)output_image_buffer,cv::Mat::AUTO_STEP);` to `cv::Mat [...]

Thx, it worked for depth_image. Is there any way to get an RGB image of the depth like the k4aviewer outputs.

Also, the output image of k4a_transformation_depth_image_to_color_camera() still looks wired.

Thx

UnaNancyOwen, 2020-03-19:
You can get depth and convert to cv::Mat as following. It is a 16bit 1channel image.
```cpp
uint8_t* buffer = k4a_image_get_buffer( depth_image );
int32_t width = k4a_image_get_width_pixels( depth_image );
int32_t height = k4a_image_get_height_pixels( depth_image );
cv::Mat depth_mat = cv::Mat( height, width, CV_16UC1, reinterpret_cast<uint16_t*>( buffer ) );
```
In addition, 16-bit image can't be displayed. You need convert to 8-bit image.
There are some ways to do this. The following is one of them.
It is convert to image of near (black) and far (white). It is 8bit 1channel image.
```cpp
// e.g. 0-5000 (16bit) -> 0-255 (8bit)
depth_mat.convertTo( depth_mat, CV_8U, 255.0 / 5000.0, 0.0 );
cv::imshow( "depth image", depth_mat );
cv::waitkey( 0 );
```

k123jack, 2020-03-23:
> You can get depth and convert to cv::Mat as following. It is a 16bit 1channel image.
> 
> ```c++
> uint8_t* buffer = k4a_image_get_buffer( depth_image );
> int32_t width = k4a_image_get_width_pixels( depth_image );
> int32_t height = k4a_image_get_height_pixels( depth_image );
> cv::Mat depth_mat = cv::Mat( height, width, CV_16UC1, reinterpret_cast<uint16_t*>( buffer ) );
> ```
> 
> In addition, 16bit image can't be displayed. You need convert to 8bit image.
> There are some ways to do this. The following is one of them.
> It is convert to image of near (black) and far (white). It is 8bit 1channel grayscale image.
> 
> ```c++
> // e.g. 0-5000 (16bit) -> 0-255 (8bit)
> depth_mat.convertTo( depth_mat, CV_8U, 255.0 / 5000.0, 0.0 );
> cv::imshow( "depth image", depth_mat );
> cv::waitkey( 0 );
> ```

Thx very much.

closed 2020-03-23 by k123jack