microsoft/Azure-Kinect-Sensor-SDK #1142: transformation_example not outputting .ply files

philianeles, 2020-03-20:
**Describe the bug**

**To Reproduce**

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**
<!- Maybe I am forgetting to specify some paths?-->

UnaNancyOwen, 2020-03-21:
It's not a bug.
The example requires command arguments to be passed at runtime.
That console shows you how to use it.
If you want to capture from AKDK device, set it up as follows.
(Please create the output directory in advance.)
```
capture C:\output
```
Please see this page how to pass command arguments on Visual Studio.
https://docs.microsoft.com/en-us/cpp/build/reference/debugging-prop-pages?view=vs-2019#command-arguments

philianeles, 2020-03-21:
ok great thank you!

closed 2020-03-21 by philianeles