microsoft/Azure-Kinect-Sensor-SDK #1144: Lost BodyTracking accuracy with Sensor Orientation 90° or 270°

JpEncausse, 2020-03-22:
**Describe the bug**
The BodyTracking with Sensor Orientation 90 or 270 is jumpy and not accurate

**To Reproduce**
1. Setup a Kinect with an Orientation at 90 or 270 
2. Add some tilt to see feet
3. Display skeleton on top of color frames

**Expected behavior**
- The body tracking should be as accurate as the regular orientation.
- The body tracking should not be jumpy (feet crossing sometimes, etc ...)

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.0.1 
 - Firmware version:  latest

**Additional context**
- Tested with an additional layer (that should not set side effect) kinect-azure (electron implementation)
- Tested in home light environment
- No CPU/GPU issues (at 30% on GTX-1080) 

I think, at 90 or 270 ° with tilt, the image magnified (bigger head in video) and the body tracking might lost accuracy: https://www.youtube.com/watch?v=vTlx23G_0cw

qm13, 2020-03-23:
What pre-processing are you doing on the IR and depth images?

JpEncausse, 2020-03-24:
Nothing, for each frame, I take the colorFrame and bodyFrame, draw the color frame then draw line on bodyFrames:

- K4A_COLOR_RESOLUTION_720P
- K4A_DEPTH_MODE_WFOV_UNBINNED
- K4A_FRAMES_PER_SECOND_15
- K4A_IMAGE_FORMAT_COLOR_BGRA32
- K4ABT_SENSOR_ORIENTATION_COUNTERCLOCKWISE90

I'll compare again between horizontal and vertical to understand if it's an orientation issue or a lightning issue (I'm at home)

yijiew, 2020-04-01:
Hi @JpEncausse thanks for reporting this issue. Did setting K4ABT_SENSOR_ORIENTATION_COUNTERCLOCKWISE90 help this issue at all? If we use the default setting and rotate the sensor 90 degrees, is it equally bad or it is worse?

JpEncausse, 2020-04-03:
Hi sorry for the delay, here is 3 videos,

- [Fitting Room - Code : 0° - Kinect 0°](https://www.youtube.com/watch?v=5ejYrREfvfo)
- [Fitting Room - Code : 0° - Kinect 90°](https://www.youtube.com/watch?v=xRJSvjmW9z8)
- [Fitting Room - Code : 90° - Kinect 90°](https://www.youtube.com/watch?v=FfMBFBjD1AU)

So with Kinect at 0 compared to 90 there is side effect like I describe it before (with the same code) I change the position of the camera a little bit lower with less "fisheye" effect.

So may be I didn't understand the orientation parameter. with Kinect at 90 the parameter 0 or 90 doesn't seem to have impact (a little may be)

I really thought the body tracking won't work at all without a 90°

yijiew, 2020-04-03:
Hi @JpEncausse , thanks for recording all these videos. You understood the parameters correctly. If you rotate the camera 90 degrees, you should set the settings to K4ABT_SENSOR_ORIENTATION_COUNTERCLOCKWISE90. If you still set the settings to 0, it might still track but the tracking reliability will be lower. You can see the legs more jittery in your second video than the third one. I guess you will see it more clearly when you tried the k4abt_simiple_3d_viewer shipped with the msi package. 

[image]

In your case, it proved that the parameter did take effect. We will further investigate the algorithm to see what improvement we can make.

JpEncausse, 2020-04-03:
Thanks ! why don't you use the internal gyroscope to understand the Kinect is at 90, 180, 270 ?

I posted [a kind of related question](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1159) because I can't figure out how to mesure precisely the lean forward angle, if I should use the orientation or do math with joints.

closed 2020-04-03 by JpEncausse

yijiew, 2020-04-03:
Great question :D We did have a discussion about using gyroscope! But we think the SDK itself is mainly a body tracking SDK. It doesn't seem to worth require all users to pass in gyroscope data to the SDK just for detecting the orientation. We decided to provide this API setting instead. If a user wants to use gyroscope to detect the sensor orientation status, they can do it by themselves and then pass in the correct orientation to the tracker.