microsoft/Azure-Kinect-Sensor-SDK #1147: Tracker.Create fails on Unity

RamType-0, 2020-03-23:
**Describe the bug**
It seems that we could not create tracker in Body Tracking SDK 1.0.1, although that was able at Version 1.0.0. 

**To Reproduce**
Steps to reproduce the behavior:
1.Setup [sample_unity_bodytracking](https://github.com/microsoft/Azure-Kinect-Samples/tree/2a1e75a0ea7a108415b39f9df66cfe09f856029d/body-tracking-samples/sample_unity_bodytracking).

2.Create empty GameObject at new scene,and attach this script.

```
using UnityEngine;
using Microsoft.Azure.Kinect.Sensor;
using Microsoft.Azure.Kinect.BodyTracking;
using System;

public class CreateTracker : MonoBehaviour
{
    // Start is called before the first frame update
    void Start()
    {
        using (Device device = Device.Open())
        {
            device.StartCameras(new Microsoft.Azure.Kinect.Sensor.DeviceConfiguration()
            {
                CameraFPS = FPS.FPS30,
                ColorResolution = ColorResolution.Off,
                DepthMode = DepthMode.NFOV_Unbinned,
                WiredSyncMode = WiredSyncMode.Standalone,
            });

            var deviceCalibration = device.GetCalibration();
            try
            {
                Tracker.Create(deviceCalibration, new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default });
            }
            catch (Exception e)
            {
                Debug.LogException(e);
                return;
            }
            Debug.Log($"Tracker has been successfully created.");

        }
    }
}
```

3.Enter Play Mode,then you could see that 

> Tracker has been successfully created.

4.Exit Play Mode,then play again.
5.Unity Editor will freeze.[(This is known issue in Version1.0.0.)](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/932)
6.Kill Unity Editor.

7.To Update BodyTrackingSDK to Version1.0.1,Download raw [nupkg](https://www.nuget.org/api/v2/package/Microsoft.Azure.Kinect.BodyTracking/1.0.1).
8.Replace libraries(at Assets/Plugins,or Project's root directory)
- k4abt.dll
- onnxruntime.dll
- dnn_model_2_0.onnx
- Microsoft.Azure.Kinect.BodyTracking.deps.json
- Microsoft.Azure.Kinect.BodyTracking.xml
- Microsoft.Azure.Kinect.BodyTracking.dll
- Microsoft.Azure.Kinect.BodyTracking.pdb

of Version 1.0.0 with these of Version 1.0.1.(Upgrading from 1.0.0 to 1.0.1 has no dependency changes.)
9.Enter PlayMode, then you could see exception log. like

> AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, Microsoft.Azure.Kinect.BodyTracking.TrackerConfiguration configuration) (at D:/a/1/s/src/csharp/Microsoft.Azure.Kinect.BodyTracking/Tracker.cs:32)
CreateTracker.Start () (at Assets/CreateTracker.cs:24)
UnityEngine.Debug:LogException(Exception)
CreateTracker:Start() (at Assets/CreateTracker.cs:28)

**Expected behavior**
Creating tracker succeeds like before upgrading to 1.0.1.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Home 
     - Version 1903 
    - Build 18362.720 
 - SDK Version:  1.0.1 
 - Firmware version:  
    - RGB camera firmware:      1.6.102
    - Depth camera firmware:    1.6.75
    - Depth config file:        6109.7
    - Audio firmware:           1.6.14
    - Build Config:             Production
    - Certificate Type:         Microsoft

- Unity Version : 2019.3.6f1

**Additional context**
Maybe this isn't a unity specific problem.

yijiew, 2020-03-23:
Thanks for reporting the issue. Could you try to generate the k4abt log file according to the instruction here: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting

yijiew, 2020-03-23:
The onnxruntime.dll should be at the same location of the k4abt.dll. Could you check whether it is the case in your project?

RamType-0, 2020-03-24:
> Thanks for reporting the issue. Could you try to generate the k4abt log file according to the instruction here: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting

I'll try it on tomorrow,but maybe this is the first time I build C++ project by myself...

> The onnxruntime.dll should be at the same location of the k4abt.dll. Could you check whether it is the case in your project?

onnxruntime.dll is placed at project's root directory,but k4abt.dll is currently placed at Assets/Plugins.
 [But the original sample unity project's readme said that they should be placed at there.](https://github.com/microsoft/Azure-Kinect-Samples/tree/2a1e75a0ea7a108415b39f9df66cfe09f856029d/body-tracking-samples/sample_unity_bodytracking)
And it works correctly on version 1.0.0.

Any way, I'll try to place them at same directory.Which directory should I place both of them?
project's root directory? or Asset/Plugins?

yijiew, 2020-03-24:
@RamType-0 The version 1.0.1 tried to fix an issue that it might acccidentally load the onnxruntime.dll at the system32 folder (The system32 onnxruntime.dll doesn't have some functionalities we need). In the new design, we require the onnxruntime.dll to be placed at the same folder as the k4abt.dll. Please place both of them at the Assets/Plugins folder and give it a try.

RamType-0, 2020-03-25:
> @RamType-0 The version 1.0.1 tried to fix an issue that it might acccidentally load the onnxruntime.dll at the system32 folder (The system32 onnxruntime.dll doesn't have some functionalities we [...]

I tried to place onnxruntime.dll at Assets/Plugins.
But creating tracker failed on both of version with exception

> AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, Microsoft.Azure.Kinect.BodyTracking.TrackerConfiguration configuration) (at D:/a/1/s/src/csharp/Microsoft.Azure.Kinect.BodyTracking/Tracker.cs:32)
CreateTracker.Start () (at Assets/CreateTracker.cs:24)
UnityEngine.Debug:LogException(Exception)
CreateTracker:Start() (at Assets/CreateTracker.cs:28)

yijiew, 2020-03-25:
Thanks for giving it a try. Could you record a k4abt log by following this instruction? https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting

RamType-0, 2020-03-27:
> Thanks for giving it a try. Could you record a k4abt log by following this instruction? https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting

I have command that 

> K4A_ENABLE_LOG_TO_A_FILE=k4a.log
K4A_LOG_LEVEL=w

but no logs appears.
Maybe this because unity doesnt use installed k4abt.
They which in their Assets/Plugins instead.

yijiew, 2020-03-27:
Thanks for giving it a try. The logging system should work as long as you are using the k4a.dll and k4abt.dll. Could you try to modify your system environmental variable to add these two line? (In addition to K4A_ENABLE_LOG_TO_A_FILE and K4A_LOG_LEVEL, you need to use separate environmental varialbe to capture K4ABT logs: K4ABT_ENABLE_LOG_TO_A_FILE and K4ABT_LOG_LEVEL) And also, the log file by default is saved to the executable folder. In your case, if you are in unity running the test scene, you might be able to find it at the folder of unity.exe. Please try to pass in a full file path instead of simply the k4a.log so that it could save to the folder you want. (e.g. use C:\Users\XXX\Desktop\k4abt.log to save it to your desktop)

RamType-0, 2020-04-02:
[image]
But no logs appears...

yijiew, 2020-04-02:
Hmm... Strange... Could you try to download the msi package from here: https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download. And run the k4abt_simple_3d_viewer.exe to see whether it works?

RamType-0, 2020-04-08:
> Hmm... Strange... Could you try to download the msi package from here: https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download. And run the k4abt_simple_3d_viewer.exe to see whether it works?

This kind of log appears.
It looks like there is no problem.
Body tracking works properly.

> 

New logging session started

[2020-04-08 10:24:00.308] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (278): ******************** Device Info ********************
[2020-04-08 10:24:00.308] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (279): K4A SDK version:     1.3.0
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (285): Serial Number:       000375694412
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (289): RGB Sensor Version:  1.6.108
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): Depth Sensor Version:1.6.79
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (295): Mic Array Version:   1.6.14
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (298): Sensor Config:       6109.7
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (301): Build type:          Release
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (306): Signature type:      MSFT
[2020-04-08 10:24:00.333] [critical] [t=25432] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (312): ****************************************************
[2020-04-08 10:24:00.701] [warning] [t=28792] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dewrapper\dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 46ms (this may be transient).

yijiew, 2020-04-08:
So does it save these log to a file or you see them on the command prompt after setting K4A_ENABLE_LOG_TO_A_FILE=k4a.log

RamType-0, 2020-04-08:
> So does it save these log to a file or you see them on the command prompt after setting K4A_ENABLE_LOG_TO_A_FILE=k4a.log

Exactly.

joel1st, 2020-04-14:
I can confirm that the project instructions are in a broken state as of v1.0.1. 
After running the `MoveLibraryFiles.bat` script it and running the project it reports `K4A_RESULT_FAILED` when creating the tracker.
To fix this you need to copy/cut the `onnxruntime.dll` into the Assets/Plugins folder.

qm13, 2020-04-14:
@joel1st can you verify that @anantley fix has fixed the issue?

RamType-0, 2020-04-15:
> @RamType-0 and @joel1st can you verify that @anantley fix has fixed the issue?

It was not perfect. I have need to place 
- cublas64_100.dll
- cudart64_100.dll

at project's root directory.

Because these DLL files are ignored in git repository,and previously placed at project's root directory,if you are upgrading from previous version,it works(which is evil).

BTW,it seems like we need to place onnxruntime.dll both in Assets/Plugins and project's root directory.
But onnxruntime.dll at project's root directory is not locked while executing body tracking.

anantley, 2020-04-15:
Hi RamType-0,  thanks for the feedback.

I also see that onnxruntime.dll is in two places. I can’t get the Tracker to open unless it is in both places.
Regarding cublas64_100.dll and cudart64_100.dll I have both of them only in the Assets/Plugins folder and it is working fine.  What could be the difference between us?  I have shut down the editor and restarted and it still works.
Best,
Angus

RamType-0, 2020-04-15:
> What could be the difference between us?

What version of Unity are you using? I'm using Unity 2019.3.6f1.

anantley, 2020-04-15:
2018.3.0f2
I will try 2019 and see if that is the diff

Angus

anantley, 2020-04-15:
Hi I tried the scene in Unity version 2019.2.6f1 and it worked fine

weichih25, 2020-04-17:
> 
> 
> > @RamType-0 and @joel1st can you verify that @anantley fix has fixed the issue?
> 
> It was not perfect. I have need to place
> 
>     * cublas64_100.dll
> 
>     * cudart64_100.dll
> 
> 
> at project's root directory instead of Assets/Plugins.
> 
> Because these DLL files are ignored in git repository,and previously placed at project's root directory,if you are upgrading from previous version,it works(which is evil).
> 
> BTW,it seems like we need to place onnxruntime.dll both in Assets/Plugins and project's root directory.
> But onnxruntime.dll at project's root directory is not locked while executing body tracking.

This solution does work for me.
Thanks :)

RamType-0, 2020-04-17:
@anantley
Does your body tracking (also) work with 

- cublas64_100.dll
- cudart64_100.dll

are in project's root directory instead of Assets/Plugins?

If it works,it seems that my solution is better.
if it doesnt work, I recommend you to place them in both directory for now(at master branch).

anantley, 2020-04-27:
@RamType-0 
Hi sorry for the delay, I am now able to recreate your issue and I will make a change to add 
cublas64_100.dll
cudart64_100.dll
to the project root.

RamType-0, 2020-04-28:
> @RamType-0 
> 
> Hi sorry for the delay, I am now able to recreate your issue and I made a pr to add 
> 
> cublas64_100.dll
> 
> cudart64_100.dll
> 
> to the project root.
> 
> 
> 
> When it is merged can you verify that this works for you?
> 
> 
> 
> Thanks
> 
> Angus
> 
> 

@anantley

Thanks!
I'll try it on when it merged.

closed 2020-05-06 by qm13

Yuki-Sato-I, 2020-06-22:
I also have a headache on this issue.

As described above,

- cublas64_100.dll
- cudart64_100.dll
- onnxruntime.dll

are in my Assets/Plugins and root directory.

but not worked;

The error  is
>AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, Microsoft.Azure.Kinect.BodyTracking.TrackerConfiguration configuration) (at <b58a402688354decb208856b8b50f97b>:0)
AzureKinect.GetDataFromKinect () (at Assets/AzureKinect.cs:77)
AzureKinect.Update () (at Assets/AzureKinect.cs:121)

My project version is
BodyTracking v 1.0.1
AzureKinectSensor v1.3.0
Unity 2018.3.12f1

Windows10 

These packages are installed by Unity Nuget (https://github.com/GlitchEnzo/NuGetForUnity)

Please help :)

BasPaap, 2021-08-03:
This still doesn't work. In my project root directory I have cublas64_11.dll, cublasLt64_11.dll, cudart64_110.dll, cudnn64_8.dll, and onnxruntime.dll. These are also in the Assets/Plugins folder. When starting the sample_unity_bodytracking sample from the Azure-Kinect-Samples repository, it still throws an exception on Tracker.Create().

KikeVilla, 2022-06-01:
Somebody found some fix for this issue?. I got the same problem.

berkan20, 2022-08-07:
I had the same issue. In addition to that, I was not able to run **Kinect Body Tracking Viewer**. I have downgraded the NVIDIA driver to 472.12. This solved the problem with **Kinect Body Tracking Viewer**. Then I have followed the steps in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1543. I managed to fix the problem with the location of the dlls.  I can run Tracker.Create() now.

- Unity Version 2020.3.31f1
- kinect body tracking sdk v1.1.2 
- windows10
- nvidia studio driver 472.12