microsoft/Azure-Kinect-Sensor-SDK #1148: Does there exist a way to overlay joint data on Kinect MKV?

ROODAY, 2020-03-23:
I have a MKV file that was passed through the [offline processor](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor), giving me a JSON file of joint data. Is there a built-in/existing method for overlaying the resultant joint data over the original MKV? We'd just like to see a visualization of how well our data is being capture, and when I tried plotting the data in matplotlib just to see how it looked it looked more like a blob of points than a human pose (do I need to some sort of projection to get the shape right?).

billpottle, 2020-03-28:
You can use the simple_3d_viewer application to get the point cloud data overlaid with the joint data. This is the main method we use to see how accurate it is.

ROODAY, 2020-03-29:
Is it a safe assumption that if simple_3d_viewer shows accurate tracking, the recorded MKVs will also be accurate? Before collecting data we usually run simple_3d_viewer first as a sanity check and everything seems fine, but we figured having a way to double check the output of the processor would be useful.

qm13, 2020-04-01:
@ROODAY the included simple_3d_viewer does not include playback. There is an open issue #848 to add playback to the viewer. The code for the viewer is in the Azure Kinect Samples repo. We would be excited to see someone from the community address this issue.
Yes, playback from a recording via the simple_3d_viewer should allow you to sanity check a recording.

billpottle, 2020-04-01:
@qm13  - I made simple modifications to simple_3d_viewer to allow it to play pre-recorded files. Do you want me to make a pull request to the Azure-Kinect-Samples repo?

qm13, 2020-04-01:
@billpottle  - Yes, please create a pull request. Much appreciated.

billpottle, 2020-04-03:
@ROODAY  - Offline processing mode is now available in simple_3d_viewer

ROODAY, 2020-04-04:
Thank you so much!

closed 2020-04-04 by ROODAY