microsoft/Azure-Kinect-Sensor-SDK #1152: How can we use Azure Kinect without Nvidia graphic card? Which Intel’s graphic card supports Azure Kinect?

ak-selfit, 2020-03-25:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

ak-selfit, 2020-03-25:
How can we use Azure Kinect without Nvidia graphic card?
Which Intel’s graphic card supports Azure Kinect?

tesych, 2020-03-31:
@ak-selfit thank you for your question. Azure Kinect Sensor SDK currently supports any graphic cards.

closed 2020-03-31 by tesych

ak-selfit, 2020-03-31:
Sorry, but I had to be more specific, I'm talking about skeleton detection which uses CUDA, Kinect V2 skeleton detection works fine without nVidia graphic card, while the Azure don't.
Please see error message we get:
[2020-03-18 13:46:59.440] [error] [t=1868] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (157): Create(). C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime::CudaCall C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::CudaCall CUDA failure 35: CUDA driver version is insufficient for CUDA runtime version ; GPU=-1307735856 ; hostname=DESKTOP-QCEJQDF ; expr=cudaSetDevice(device_id_);

clpacheco90, 2020-06-30:
Sorry but AK Bodytracking only supports NVIDIA's GPU: https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements#body-tracking-host-pc-hardware-requirements

ak-selfit, 2020-07-01:
Thanks for your answer.
Do you plan in future to enhance the support to non-NVIDIA GPU?