microsoft/Azure-Kinect-Sensor-SDK #1153: Upgrade fails

hrsFire, 2020-03-27:
**Describe the bug**
I've tried to update my packages on Ubuntu 18.04 Bionic Beaver and recognized an annoying bug which caused the updating process to fail and to leave a broken system.

**To Reproduce**
1. Sensor SDK (1.3) and Body SDK (1.0) are already installed
2. Start the update process by using "sudo apt update && sudo apt upgrade -y"
4. See error:
Unpacking libk4a1.3 (1.3.0) ...
dpkg: error processing archive /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb
 (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libk4a.so', which is also in pac
kage libk4a1.4 1.4.0
Errors were encountered while processing:
 /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

**Expected behavior**
A clean install

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 Bionic Beaver
 - SDK Version:  1.0
 - Packages: libk4abt1.0

**Additional context**
The package "k4a-tools" depends on "libk4a1.4" but the package "libk4abt1.0" depends on "libk4a1.3". There are no updated versions available for the package "libk4abt".

wes-b, 2020-03-27:
The most recent version of body tracking is tied to K4A version 1.3. So you need to uninstall K4A 1.4 until a new BT version comes out.

closed 2020-03-27 by wes-b

hrsFire, 2020-03-27:
That's ridiculous. This means in fact that i can't use "k4a-tools" side by side with "libk4abt1.0-dev". Are you serious? Good user experience looks different.

RoseFlunder, 2020-03-27:
You can still use k4a-tools version 1.3.0
Remove your current sdk 1.4.0, install sdk 1.3.0 and then install k4a tools with:
sudo apt install k4a-tools=1.3.0

hrsFire, 2020-03-28:
Thanks! That works

hrsFire, 2020-04-01:
Works only until the next system update. It's required to hold the package to avoid a broken installation.
echo "k4a-tools hold" | sudo dpkg --set-selections

RoseFlunder, 2020-04-01:
Ah yes thats true, I forgot to mention that you should not try to upgrade it again.
I did it this way which has the same effect:
sudo apt-mark hold k4a-tools