microsoft/Azure-Kinect-Sensor-SDK #1155: transformation_example | "k4a::k4a"

alitokur, 2020-04-01:
I have a problem in building "transformation_example". I am using cmake to build but it can't find target  libraries.

Error Lines : 

-  Target "transformation_example" links to target "k4a::k4a" but the target was not found.  Perhaps a find_package() call is missing for an IMPORTED  target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

- Target "transformation_example" links to target "k4a::k4arecord" but the target was not found.  Perhaps a find_package() call is missing for an  IMPORTED target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it

- Target "transformation_example" links to target "libjpeg-turbo::libjpeg-turbo" but the target was not found.  Perhaps a  find_package() call is missing for an IMPORTED target, or an ALIAS target is missing? This warning is for project developers.  Use -Wno-dev to suppress it.

After this steps, I still opened the solution window.
 [image] 
As you can see, header files are missing. After using nugget, and adding turbojpeh include.dir, then i think, have no problem with them. 

[image]

And now, when i try to build, i see the following error:
[image]

I'm sorry, not familiar with vs studio. What should I do?

 - OS : Windows 10 
 - SDK Version:  1.4.0 
 - Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7

tkircher, 2020-04-07:
On Linux, in order to get some of these examples to build, I had to manually edit the link.txt files to change -lk4a::k4a to -lk4a, etc. because k4a::k4a is of course not the name of any library. Going to try to track down the source of the issue sometime.

alitokur, 2020-04-13:
thank u @tkircher , i'm not sure, but when I added ` #pragma comment(lib, "k4a.lib")` , the problem has been fixed.

closed 2020-04-13 by alitokur