microsoft/Azure-Kinect-Sensor-SDK #1156: Invalidaton of depth pixels in edges

sanmaebal4, 2020-04-01:
I am using the Azure Kinect with the NFOV Unbbined depth mode to detect a chess board. I make a capture of the chess board and then this capture is transformed in a point cloud through a programme designed in Visual Studio in C/C++. The point cloud is saved in a PLY file and I can open it in a programme named CloudCompare which allows me to make measures from the point cloud of the chess board.
Which I observed (you can see it in the images below) is that in the edges of the chess board, a lot of pixels dissapear. So I made some tests and I could see that the greater the distance is, the greater the loss of pixels appear. The problem started having the chessboard 2 meters of distance from the camera, and for 3, 4 or 5 meters the problem increase.
At the same time, I discovered that if there is a wall just behind the chessboard the pixels didn't dissapear. Is like the camera has a filter which invalidate pixels that are in an edge of an object and it has a lot of distance from his background.

What I want to know if it is possible to remove this filter or include something in my Visual Studio programme to don't lose this dissapeared pixels.
Below there are some screenshots of my problem:

- First photo: Chess board with a wall behind don't lose pixels (CloudCompare view).
[image]

- Second photo: Chess board lose pixels speciallly from the top edge, when it is separated from the wall (CloudCompare view).
[image]

- Third and fourth photo: Detailed view of the losed pixels (CloudCompare view).
[image]
[image]

- Fifth, sixth and seventh photo: There is another example of my problem, but now you can see that the problem is the same in the Azure Kinect Viewer and CloudCompare, for this reason I think that this is not a problem of my Visual Studio programme or CloudCompare, because the Viewer shows the same problem.
[image]
[image]
[image]

sanmaebal4, 2020-04-10:
I saw that you put the label "More info needed". Could you tell me which information you need?

qm13, 2020-08-10:
@sanmaebal4 looking at the images we believe that the edge does not have enough SNR to produce reliable depth and have been eliminated by the depth processor.

closed 2021-04-14 by qm13