microsoft/Azure-Kinect-Sensor-SDK #1158: Unity BodyTracking Sample Quaternion is broken

RamType-0, 2020-04-02:
**Describe the bug**

Unity Sample's Quaternion coordinate has some inverted,or rotating axis.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Home 
     - Version 1903 
    - Build 18362.720 
 - SDK Version:  1.0.0 (I could not create tracker in 1.0.1) 
 - Firmware version:  
    - RGB camera firmware:      1.6.102
    - Depth camera firmware:    1.6.75
    - Depth config file:        6109.7
    - Audio firmware:           1.6.14
    - Build Config:             Production
    - Certificate Type:         Microsoft

- Unity Version : 2019.3.6f1
**Additional context**
I think Azure kinect's coordinate system has so many problems.(e.g. inverted y axis)

anantley, 2020-04-07:
Hi I took a look at this and the rotations are correct.  They follow the documentation "All joint coordinate systems are absolute coordinate systems relative to the depth camera 3D coordinate system".  The scene hierarchy of the figure in the app is the same, ie each joint has the body root as its parent.  If you want to find the relative joint rotations, it is just a matter of multiplying the joint by the inverse rotation of the parent, using the parent joint map.  This map is defined in the TrackerHandler.cs file.  The joint hierarchy is also listed in the Azure Kinect Documents.

RamType-0, 2020-04-07:
> Hi I took a look at this and the rotations are correct. They follow the documentation "All joint coordinate systems are absolute coordinate systems relative to the depth camera 3D coordinate [...]

I know that every positions and rotations in joint is absolute.(It is relative position to azure kinect depth camera coordinate system)

First of all,this problem starts with difference between azure kinect coordinate system and unity coordinate system.(XYZ axis direction,and distance unit)

In unity body tracking sample, 
positions were marshaled almost properly.
(Y coordinate is inverted. 
But I think units conversion from millimeter to meter is also needed.)

``` C#
Vector3 jointPos = new Vector3(skeleton.JointPositions3D[jointNum].X, -skeleton.JointPositions3D[jointNum].Y, skeleton.JointPositions3D[jointNum].Z);
```

But quaternions were not. They were invalidly-blitted.

``` C#
Quaternion jointRot = new Quaternion(skeleton.JointRotations[jointNum].X, skeleton.JointRotations[jointNum].Y, skeleton.JointRotations[jointNum].Z, skeleton.JointRotations[jointNum].W);
```

knat, 2020-04-07:
```C#
Quaternion jointRot = new Quaternion(
  skeleton.JointRotations[jointNum].X,
  -skeleton.JointRotations[jointNum].Y, 
  skeleton.JointRotations[jointNum].Z,
  -skeleton.JointRotations[jointNum].W);
```

RamType-0, 2020-04-08:
> ```cs
> Quaternion jointRot = new Quaternion(
>   skeleton.JointRotations[jointNum].X,
>   -skeleton.JointRotations[jointNum].Y, 
>   skeleton.JointRotations[jointNum].Z,
>   -skeleton.JointRotations[jointNum].W);
> ```

Thanks!

It looks like

``` C#
Quaternion jointRot = new Quaternion(skeleton.JointRotations[jointNum].X, -skeleton.JointRotations[jointNum].Y, skeleton.JointRotations[jointNum].Z, -skeleton.JointRotations[jointNum].W) * Quaternion.Inverse(Quaternion.Euler(0, -90, 90));
```

is correct.

anantley, 2020-04-08:
Hi sorry @RamType-0 I think I misunderstood the problem.  Each of the joints has its own coordinate system according to the documentation.  This means that each joint rotation has a sort of pre-rotation followed by a delta rotation that is the current reading from the Kinect.  What I found yesterday was that the delta rotation displayed correctly in Unity.  To use them you need to perform a change of basis for each joint if you want to get the rotation in the standard basis.  
CurrentRawKinectReading = PreRotation * DeltaRotation
ReadingInAbsoluteWorld = PreRotation * DeltaRotation * PreRotationInverse 

That will give you the current absolute rotation of the joint, where the start point will be the skeleton in the t-pose.  In other words the t-pose means every joint has the identity rotation.

I will push a pr for this change to the code  tomorrow.  It will also have a function that gives you the relative rotation for each joint in the standard basis.

My apologies if I have still not understood what you are looking for. In my display of the rotations the solution you propose above gives you a mirrored effect.  Is that what you are looking for?

RamType-0, 2020-04-08:
> What I found yesterday was that the delta rotation displayed correctly in Unity.

> My apologies if I have still not understood what you are looking for. In my display of the rotations the solution you propose above gives you a mirrored effect. Is that what you are looking for?

I got flipped delta rotation with current unity sample.
Are you misunderstanding your back as chest?
When we facing with azure kinect,our forward direction is -Z.

anantley, 2020-04-16:
FYI the pr has been merged that gives you each joint in camera sensor space with each bone being in t-pose with the quat (0.0,0.0,0.0,1.0) xyzw.

qm13, 2020-04-25:
Fixed

closed 2020-04-25 by qm13