microsoft/Azure-Kinect-Sensor-SDK #1159: Best Practices: Body Tracking Orientation vs Gyroscope

JpEncausse, 2020-04-03:
**Describe the bug**

According to the documentation the [Body Tracking is computed based on depth camera system](https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints). Based on the [camera coordinates system](https://docs.microsoft.com/en-us/azure/kinect-dk/coordinate-systems) but not the Gyroscope.

Standing in front of the camera, how would you compute the body angle of PELVIS / SPINE / LEGS ? How much body "lean"

- Should I use the PELVIS orientation ? 
    - Is it accurate enough ?
    - Is it the orientation from other joints (aka LEGS) or floor (using Gyroscope)
- or should I calculate the normal of HIP / SPINE ? 

We try option 2 but results are not very accurate, so I wondering how would you compute that angle especially if the body is standing behind a desk and we don't see very well the legs.

**To Reproduce**
1. Stand in front of the camera
2. Lean forward
3. Calculate the angle of the PELVIS / BACK / LEGS (or FLOOR)

[image]

**Expected behavior**

Improve documentation to explain
- How body orientation is computed (does it use Gyroscope)
- How to compute body angles with or without Gyroscope (aka floor)

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.3.1
 - Firmware version:  latest

**Additional context**
It's not a bug, but more a clarification required in order to be able to compute correctly the

knat, 2020-04-08:
Two joints constitute a bone, a bone is a vector, In right-handed coordinate(+x: right, +y: up, +z: from screen to user), we can simply assume "How much body lean" = the quaternion of the PELVIS-SPINE_NAVAL vector and the up vector(0,1,0). C# code:
```C#
using Microsoft.Azure.Kinect.BodyTracking;
using System;
using System.Diagnostics;
using System.Numerics;

namespace Test
{
    public static class Class1
    {
        public static Vector3 ToRightHanded(this Vector3 kinectVec) => new Vector3(kinectVec.X, -kinectVec.Y, -kinectVec.Z);

        /*
        https://stackoverflow.com/questions/1171849/finding-quaternion-representing-the-rotation-from-one-vector-to-another
        Quaternion q;
        vector a = crossproduct(v1, v2);
        q.xyz = a;
        q.w = sqrt((v1.Length ^ 2) * (v2.Length ^ 2)) + dotproduct(v1, v2);       
        */
        public static Quaternion GetQuaternionOf(Vector3 v1, Vector3 v2)
        {
            v1 = Vector3.Normalize(v1);
            v2 = Vector3.Normalize(v2);
            Vector3 a = Vector3.Cross(v1, v2);
            return Quaternion.Normalize(new Quaternion(a, (float)(Math.Sqrt((v1.Length() * v1.Length()) * (v2.Length() * v2.Length()))
                + Vector3.Dot(v1, v2))));
        }

        //https://stackoverflow.com/questions/12088610/conversion-between-euler-quaternion-like-in-unity3d-engine
        public static void QuaternionToEulerAngles(float x, float y, float z, float w,
            out float rx, out float ry, out float rz)
        {
            float test = x * w - y * z;

            if (test > 0.4995f)// * unit)
            { // singularity at north pole
                rx = (float)Math.PI / 2;
                ry = 2f * (float)Math.Atan2(y, x);
                rz = 0;
            }
            else if (test < -0.4995f)// * unit)
            { // singularity at south pole
                rx = -(float)Math.PI / 2;
                ry = -2f * (float)Math.Atan2(y, x);
                rz = 0;
            }
            else
            {
                float sqx = x * x;
                float sqy = y * y;
                float sqz = z * z;

                rx = (float)Math.Asin(2f * (w * x - y * z));
                ry = (float)Math.Atan2(2f * w * y + 2f * z * x, 1 - 2f * (sqx + sqy));
                rz = (float)Math.Atan2(2f * w * z + 2f * x * y, 1 - 2f * (sqz + sqx));
            }
            NormalizeRadAngle(ref rx);
            NormalizeRadAngle(ref ry);
            NormalizeRadAngle(ref rz);
        }
        public const float TwoPi = 6.28318548F;
        public static void NormalizeRadAngle(ref float f)
        {
            while (f < 0) f += TwoPi;
            while (f > TwoPi) f -= TwoPi;
        }

        public static void Test(Skeleton skeleton)
        {
            var pelvisPos = skeleton.GetJoint(JointId.Pelvis).Position.ToRightHanded();
            var spineNavelPos = skeleton.GetJoint(JointId.SpineNavel).Position.ToRightHanded();
            var q = GetQuaternionOf(spineNavelPos - pelvisPos, Vector3.UnitY);
            QuaternionToEulerAngles(q.X, q.Y, q.Z, q.W, out var rx, out var ry, out var rz);
            Debug.WriteLine("rx: {0}, ry: {1}, rz: {2}", rx, ry, rz);
        }
    }
}
```

knat, 2020-04-08:
Depending on how to define the "body lean", maybe PELVIS-SPINE_CHEST vector is more accurate.

JpEncausse, 2020-04-09:
@knat thanks for the explanation, there is something I missed

You assume the up vector(0,1,0) is the top but in real life the camera itself is not perfectly straight forward and may have 45°angle : 

1. does the bone vector is wrongly oriented by the camera orientation (and so we are comparing 2 referential: the bone vector in the camera referential and (0.1.0) from the real world referential)
2. or does the bone vector is "fixed" by the DNN Model or the Gyroscope ?

According to the Kinect documentation I thought it was 1. because everything is calculated from the camera position.

knat, 2020-04-09:
Maybe you should replace the up vector(`Vector3.UnitY` in the code) with the gyroscope value.

immerse-ar, 2020-04-09:
@JpEncausse Can you let me know if this works? Also running into a similar issue. Placing cameras at angles (rotated 45 degrees like the 2nd picture above) causes issues with using the joint position data

JpEncausse, 2020-04-10:
@immerse-ar so the correct value return by the kinect is the image on the left. It works correctly using  vectors in the camera refereal but using vector outside of the camera (the roof) require to use gyroscope or other tricks.

The X,Y,Z position seems Ok (depends of the acuracy you need) but animating a puppet is "hard" because you have to translate orientation relative to each joint (but it's another story)

immerse-ar, 2020-04-10:
@JpEncausse thanks for the response! Since it is exactly the same as your issue, I created a separate issue (/request for documentation) on how to do this here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1168 

I've been trying to translate orientation using IMU data from the kinect and equations provided in this link: https://www.cs.helsinki.fi/group/goa/mallinnus/3dtransf/3drot.html but still can't seem to translate it correctly.

Can you offer any advice or links on how to "translate orientation to each joint"? (preferably on the other issue so I don't get in the way of them helping on your specific issue)

qm13, 2020-04-17:
@immerse-ar have you looked at this sample https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/floor_detector_sample. It shows how to obtain the gravity vector. You can then derive the rotation to align the gravity vector to one of the axes of the desired world reference (e.g. [0,1,0] for y-axis. There is a good reference for this step here https://gist.github.com/kevinmoran/b45980723e53edeb8a5a43c49f134724
Then you can apply this rotation to all the joint transforms to obtain the skeleton in a gravity aligned frame of reference.

closed 2020-05-06 by qm13