microsoft/Azure-Kinect-Sensor-SDK #1163: zero length normals

philianeles, 2020-04-07:
Hi, I have this sample point cloud collected with Azure Kinect DK: https://www.dropbox.com/s/9qjmqt98n4fpse0/0.1ms_to_0.05ms_Ed_500mm.ply?dl=0

I used the `K4A.sln` --> `transformation_example` to create the .ply sample.

I seem to have zero length normals, or normals with infinity/undefined coefficients.
Any clues or settings need to be changed to avoid Nan normals?

Thanks!

RamType-0, 2020-04-07:
Although you can see "it" from color camera,there is a case you will not see "it" from depth camera.
While transformation facing this problem,transformation treat "it" as zero position.

And then,We could not compute normal from point cloud which all of them has exactly same position.

philianeles, 2020-04-08:
Is there a way to retrieve and save point cloud (.ply for instance) and have the normals as well?

RamType-0, 2020-04-08:
> Is there a way to retrieve and save point cloud (.ply for instance) and have the normals as well?

It seems like completely the other question,and I don't know the answer.
I recommend you to create another issue.

rabbitdaxi, 2020-04-21:
The transformation_example does not compute surface normal, it only computes point cloud xyz with color (the example is just leveraging ply as a simple format to allow user visualize the point cloud in MeshLab or other tools). To compute the normal from a depth point cloud, there are many ways to do that, naïve way is estimating normal from neighbor pixels but need to be careful with the depth discontinuity area. Or you can read some more papers with advanced way. Feel free to contribute to the example with normal computation :)

closed 2020-07-11 by qm13